PowerBroker2/SerialTransfer #10: Getting 0s at Rx in Arduino Uno when connected via software Serial

antno1000, 2019-12-25:
**Describe the bug**
Running the attached Examples Arduino file namely **rx_multiple_objects.ino** and **tx_multiple_objects.ino** gives 0s instead of struct and array's data at receiving end.
the problem seems to be with  myTransfer.txObj() function as the data sent directly via myTransfer.txBuff[0]='h';  seems to appear  just fine.

the example was run without any modifications other than using softserial instead of Serial1.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Examples and upload tx_multiple_objects and rx_multiple_objects to arduino'
2. Sample files are attached
[Serial.zip](https://github.com/PowerBroker2/SerialTransfer/files/4000616/Serial.zip)

**Expected behavior**
Data from array should have appeared on the serial monitor of Arduino.

**Screenshots**
[image]

 - OS: Windows 10
 - Arduino IDE [1.8.9]

GranTesoro86, 2019-12-27:
@PowerBroker2 
I had the same problem, I used nodemcu and due , then I hope to make it easier to transfer strings

PowerBroker2, 2019-12-27:
I haven't looked into it in depth yet (soon), but one thing to note is that you may get errors if the endian-ness or processor is different (i.e. 8-bit processor vs a 32 bit processor)

PowerBroker2, 2019-12-27:
So it looks like the bytes **are correctly transferring**, but the passed objects are **not being correctly updated**.

For debugging purposes, I used the code below to print the packet payload bytes individually and got the following results:

Debugging Code:
```#include "SerialTransfer.h"

SerialTransfer myTransfer;

struct STRUCT {
  char z;
  float y;
} testStruct;

uint16_t arr[3] = {};

void setup()
{
  Serial.begin(115200);
  Serial1.begin(115200);
  myTransfer.begin(Serial1);
}

void loop()
{
  if(myTransfer.available())
  {
    // use this variable to keep track of how many
    // bytes we've processed from the receive buffer
    uint16_t recSize = 0;

    for(byte i = 0; i < myTransfer.bytesRead; i++)
    {
      Serial.print(myTransfer.rxBuff[i]);
      Serial.print(' ');
    }
    Serial.println();
  }
  else if(myTransfer.status < 0)
  {
    Serial.print("ERROR: ");
    Serial.println(myTransfer.status);
  }
}
```

Results:
[image]

I will take a deeper look into why the passed objects aren't updated...

PowerBroker2, 2019-12-27:
Ok, found the issue and fixed it. Using two Mega MCUs I now have the following (successful) results:

[image]

PowerBroker2, 2019-12-27:
Thank you both for uncovering and reporting the bug! Please let me know if you have any other problems or improvements.

closed 2019-12-27 by PowerBroker2

antno1000, 2019-12-27:
Thanks for fixing it, I'll give it a try!!