PowerBroker2/SerialTransfer #11: Unable to transmit string correctly

GranTesoro86, 2019-12-27:
**Describe the bug**
I tested with nodemcu and due , examples work fine 
So I tried to transfer the string , 

**To Reproduce**
Steps to reproduce the behavior:
1. Download [them](https://cdn.discordapp.com/attachments/627701490173345811/660058877852909578/code.rar) and upload to arduino
2. See error ' ? '

**Expected behavior**
rx should show the string , but I get a bunch of  ' ? '

**Screenshots**
[image]

 - Board: NodeMCU and Arduino due
 - Arduino IDE [1.8.10]

PowerBroker2, 2019-12-27:
Using "S"trings isn't the best idea to use with Arduinos because they can lead to memory issues and whatnot.

Regardless, can you post/link your code?

PowerBroker2, 2019-12-27:
I've decided that the use of Strings within structs will not be supported because of how Strings work and are stored in memory. As a fix, you can change your struct members from Strings to char arrays and use the str.c_str() member function to convert your strings. See this link: [http://www.cplusplus.com/reference/string/string/c_str/](url)

PowerBroker2, 2019-12-27:
I may soon add a new function to transfer individual Strings - just not within structs.

GranTesoro86, 2019-12-28:
I tried to transmit using char array, but I couldn't receive it 
[code.zip](https://github.com/PowerBroker2/SerialTransfer/files/4006830/code.zip)
[image]

PowerBroker2, 2019-12-28:
Your code won't compile - not even sure how you were able to test it...
Error:

> error: flexible array member

You will need to specify a max array length for the member char arrays. I tested your code specifying max array length of 10 for each and got successful results.

Here's what I changed the struct definition to (in both RX and TX sketches):
```struct STRUCT {
  char room_text[10];
  char user_text[10];
  char status_text[10];
} testStruct;```

closed 2019-12-28 by PowerBroker2