PowerBroker2/SerialTransfer #12: SerialTransfer <variable> breaks ability to read SD card.

SSolheim2000, 2020-03-08:
Here's my code: [LED_Sequencer_Working_Copy - Copy.txt](https://github.com/PowerBroker2/SerialTransfer/files/4302865/LED_Sequencer_Working_Copy.-.Copy.txt)

**Describe the bug**
When I activate (uncomment) the line "SerialTransfer myTransfer;", my program can no longer read the DATA.txt file located on the SD card.

**To Reproduce**
Steps to reproduce the behavior:
1. I've added in the code from the "tx_simple" under the examples folder. (BTW, I've tested the tx_simple/rx_simple on their own and they work perfectly).
2. I'm using this example for the SD card shield's pinouts: https://www.mschoeffler.de/2017/02/22/how-to-use-the-microsd-card-adapter-with-the-arduino-uno/
3. With all of the added code from example file commented out, the program runs fine.
4. Uncommenting the "#include "SerialTransfer.h"" and it still runs fine.
5. Uncommenting the  "SerialTransfer myTransfer;" line and my code can no longer read the Data.txt file on my SD card.

**Expected behavior**
For SerialTransfer.h to be used in conjunction with my SD card reader.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Win 10

**Smartphone (please complete the following information):**
N/A

**Additional context**
Add any other context about the problem here.

PowerBroker2, 2020-03-08:
> Uncommenting the "SerialTransfer myTransfer;" line and my code can no longer read the Data.txt file on my SD card.

What exactly do you mean? How do you know it's not reading the file?

Also, what Arduino are you using and are you using hardware or softwareserial ports?

PowerBroker2, 2020-03-08:
Also, what Arduino are you using?

SSolheim2000, 2020-03-08:
I'm using an UNO R3.
When I have the line commented out, I get an error message in the Serial Monitor:

[image]

  // Setup for the Serial Transfer module
    #include <SerialTransfer.h>
    //SerialTransfer myTransfer;

SSolheim2000, 2020-03-08:
When it's commented out, I get the expected data in the Serial Monitor:
[image]

// Setup for the Serial Transfer module
    #include <SerialTransfer.h>
    //SerialTransfer myTransfer;

PowerBroker2, 2020-03-08:
There's really no reason the library is causing an issue. I've been able to use the library successfully with an SD card [(see here)](https://github.com/PowerBroker2/RND_Flight_Code/blob/master/Data_Logger/Data_Logger.ino).

The only thing I can think of is that you're using ~73% of the Uno's dynamic memory. Even then, I'd only get worried once it hits 90%, so idk.

I suggest getting something like a Mega or (better still) a Teensy 3.5/3.6. They have more memory and hardware serial ports and the Teensy specifically has a ton more processing power.

SSolheim2000, 2020-03-08:
I changed all my "int" variables to "byte" and now it's working fine. Thanks for the good lead and quick responses!!

closed 2020-03-08 by SSolheim2000