PowerBroker2/SerialTransfer #15: RX <> TX forwarding

Tomblarom, 2020-04-04:
Sorry, I found no way to write you an email. I'm currently working on reading the communication between controller and display of my bafang ebike. The official Arudino commands are way to slow. Maybe the only solution is to directly program in C.. Have a look at my code. Am I doing anything wrong? I tried removing / playing with the delay timings. Nothing helped. The display of my ebike has communication issues and is giving me "Error Code: 30".

[Using Robodyn Mega 2560 PRO (Embed)](https://robotdyn.com/mega-2560-pro-embed-ch340g-atmega2560-16au.html)

```
#include "SerialTransfer.h"
#include <ST_CRC.h>

SerialTransfer serial1;
SerialTransfer serial2;

void setup() {
  Serial1.begin(1200);    //BBSHD controller    @RX18 TX19
  Serial2.begin(1200);    //Display DP-C18      @TX16 RX17

  serial1.begin(Serial1);
  serial2.begin(Serial2);
}

void loop() {
  readDisplay();
  readController();
}

void readDisplay()  {
  if (serial2.available()) {
    for(byte i = 0; i < serial2.bytesRead; i++) {
      serial1.txBuff[i] = serial2.rxBuff[i];
    }
    serial1.sendData(serial2.bytesRead);
	delay(100);
  }
}

void readController() {
  if (serial1.available()) {
    for(byte i = 0; i < serial1.bytesRead; i++) {
      serial2.txBuff[i] = serial1.rxBuff[i];
    }
    serial2.sendData(serial1.bytesRead);
	delay(100);
  }
}
```

PowerBroker2, 2020-04-04:
I don't have enough context about your project to understand what you're trying to do and how. SerialTransfer.h is for Arduino <--> Arduino or Arduino <--> PC (Python) communication only. Do you have an Arduino controlling the bike? If you want to do serial port forwarding, you don't need the lib - but if you're only doing port forwarding, why have the Arduino in the first place?

Tomblarom, 2020-04-04:
[This is the project](https://endless-sphere.com/forums/viewtopic.php?t=94850) I'm working on. My intention in using your lib, was to speed up the communication, if the arduino is in between. The Arduino sits in between controller and display, sniffing uart packages and reprogramming the controller if needed.

Even with this simple code I posted earlier, there's a huge delay of 2-3 seconds, in updating the displays speedometer. Do you have another idea, before I start learning pure C? :)

PowerBroker2, 2020-04-04:
You don't need "pure C" - all of the delay is most likely happening in the display itself. Since neither the bike controller nor the display know how to handle SerialTransfer packets, you can't use it to do port forwarding in this case. You can do a simple while loop like this:

```C++
while (serial2.available())
{
  char readChar = Serial2.read();
  Serial1.write(readChar);
}
```

If you have any more questions on how to do this, please ask in [the Arduino forum](https://forum.arduino.cc/) and somebody will help. Since SerialTransfer.h can't help here and there is no library bug reported, I'm going to go ahead and close the issue.

Thank you for your interest in the library and if you **find any library bugs** or have any problems with using the library for **inter-Arduino communication**, feel free to open a new issue.

closed 2020-04-04 by PowerBroker2