PowerBroker2/SerialTransfer #17: SerialTransfer conflicts with the AccelStepper library

qinglong1224, 2020-04-15:
When configuring the send / receive structure in the loop, the Stepper runs incorrectly. Comment out that the send and receive Stepper works well.

PowerBroker2, 2020-04-15:
You've provided incredibly little context to this issue and from what little _was_ provided it sounds more like a user error than library issue. If you want me to look into it, please post a clear description of your project, your entire code, full schematic, any error printouts, and what exactly is going wrong with your project.

qinglong1224, 2020-04-15:
[image]

qinglong1224, 2020-04-15:
There is no problem in the compilation process. In controlling the stepper, the speed and the number of steps are wrong. Comment out that rx tx stepper runs correctly

qinglong1224, 2020-04-15:
[image]
[image]
[image]
[image]

qinglong1224, 2020-04-15:
This is my program. I want to use it to control 42 stepper motors. During the test, the stepper is not running correctly. There is no problem with the compilation. The stepper is also running, but it is not running correctly. The comment sending and receiving stepper runs correctly

qinglong1224, 2020-04-15:
stepping motor(42mm). Throughout the test, the stepping motor still runs but it runs wrongly,without any problem in compiling.
Comment "send"and "receive",the stepping motor works fine

PowerBroker2, 2020-04-15:
When posting code in GitHub issues, [please use correct markdown](https://help.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks) instead of screenshots. Please post your code correctly so I can take a look at it.

qinglong1224, 2020-04-15:
[receive.zip](https://github.com/PowerBroker2/SerialTransfer/files/4484135/receive.zip)

qinglong1224, 2020-04-15:
```
#include <SerialTransfer.h>
#include <ST_CRC.h>
#include <AccelStepper.h>
#include <MultiStepper.h>

SerialTransfer myTransfer;

unsigned long lastReceiveTime = 0;
unsigned long currentTime = 0;

struct STRUCT {
	byte button;
} testStruct;

struct STR {
	byte XW;
} test;

#define xdirPin 2
#define xstepPin 3
AccelStepper stepper1(1,xstepPin,xdirPin);

#define Button 4

void setup()
{
	pinMode(Button, INPUT_PULLUP);
	pinMode(xstepPin,OUTPUT);
	pinMode(xdirPin,OUTPUT);
	
	stepper1.setMaxSpeed(500.0);
	stepper1.setAcceleration(5.0);
	
	Serial.begin(9600);
	myTransfer.begin(Serial);
	
	receive();
	send();
}

void loop()
{
	test.XW = digitalRead(Button);
	
	myTransfer.txObj(test, sizeof(test));
	myTransfer.sendData(sizeof(test));
	delay(100);
	
	if(myTransfer.available())
	{
		uint16_t recSize = 0;
		myTransfer.rxObj(testStruct, sizeof(testStruct), recSize);
		recSize += sizeof(testStruct);
		lastReceiveTime = millis();
	}
	currentTime = millis();
	if ( currentTime - lastReceiveTime > 1000) {
		receive();
	}
	Key();
}

void Key()

{
	if (testStruct.button == 1)
	{
		stepper1.setSpeed(0);
	}
	if (testStruct.button == 0&&test.XW == 1)
	{
		stepper1.setSpeed(100);
	}
	stepper1.runSpeed();
}

void send()
{
	test.XW = 1;
}

void receive()
{
	testStruct.button = 1;
}
```

PowerBroker2, 2020-04-15:
Thank you for the code in tags. What is your Arduino attempting to communicate with? Is it another Arduino? Is it a Python program? Is it user input from the Serial Monitor?

qinglong1224, 2020-04-15:
arduino uno

PowerBroker2, 2020-04-16:
Can you post the Uno's code AND a schematic? How are the Arduinos and motors wired?

So far I don't see anything that suggests the library is at fault here.

qinglong1224, 2020-04-16:
The TB6600 used by the driver has a common anode, pin 2 is connected to Dir-, pin 3 is connected to step-, you can directly put the motor running code into the loop, and the motor runs incorrectly.
The motor runs correctly without using the SerialTransfer library.

Stepper motor operation error:
#include <SerialTransfer.h>
#include <ST_CRC.h>
#include <AccelStepper.h>
#include <MultiStepper.h>

SerialTransfer myTransfer;

unsigned long lastReceiveTime = 0;
unsigned long currentTime = 0;

struct STRUCT {
	byte button;
} testStruct;

struct STR {
	byte XW;
} test;

#define xdirPin 2
#define xstepPin 3   //
AccelStepper stepper1(1,xstepPin,xdirPin);

void setup()
{
	pinMode(xstepPin,OUTPUT);
	pinMode(xdirPin,OUTPUT);
	
	stepper1.setMaxSpeed(500.0);
	stepper1.setAcceleration(5.0);
	
	Serial.begin(9600);
	myTransfer.begin(Serial);
	
	receive();
	send();
}

void loop()
{
	myTransfer.txObj(test, sizeof(test));
	myTransfer.sendData(sizeof(test));
	delay(100);
	
	if(myTransfer.available())
	{
		uint16_t recSize = 0;
		myTransfer.rxObj(testStruct, sizeof(testStruct), recSize);
		recSize += sizeof(testStruct);
		lastReceiveTime = millis();
	}
	currentTime = millis();
	if ( currentTime - lastReceiveTime > 1000) {
		receive();
	}
		{
		stepper1.setSpeed(100);
	}
	stepper1.runSpeed();
	
}

void send()
{
	test.XW = 1;
}

void receive()
{
	testStruct.button = 1;
}

Stepper motor runs correctly，:

#include <SerialTransfer.h>
#include <ST_CRC.h>
#include <AccelStepper.h>
#include <MultiStepper.h>

SerialTransfer myTransfer;

unsigned long lastReceiveTime = 0;
unsigned long currentTime = 0;

struct STRUCT {
	byte button;
} testStruct;

struct STR {
	byte XW;
} test;

#define xdirPin 2
#define xstepPin 3   //
AccelStepper stepper1(1,xstepPin,xdirPin);

void setup()
{
	pinMode(xstepPin,OUTPUT);
	pinMode(xdirPin,OUTPUT);
	
	stepper1.setMaxSpeed(500.0);
	stepper1.setAcceleration(5.0);
	
	Serial.begin(9600);
	myTransfer.begin(Serial);
	
	receive();
	send();
}

void loop()
{
	//myTransfer.txObj(test, sizeof(test));
	// myTransfer.sendData(sizeof(test));
	// delay(100);
	
	if(myTransfer.available())
	{
		uint16_t recSize = 0;
		myTransfer.rxObj(testStruct, sizeof(testStruct), recSize);
		recSize += sizeof(testStruct);
		lastReceiveTime = millis();
	}
	currentTime = millis();
	if ( currentTime - lastReceiveTime > 1000) {
		receive();
	}
		{
		stepper1.setSpeed(100);
	}
	stepper1.runSpeed();
	
}

void send()
{
	test.XW = 1;
}

void receive()
{
	testStruct.button = 1;
}

PowerBroker2, 2020-04-16:
Not a library issue - user error. You haven't proven the library is broken and I'm tired of dealing with the unprofessional and lazy posts you've provided in this issue.

If you need help with your project, please post a question in the [Arduino forum](https://forum.arduino.cc/)

closed 2020-04-16 by PowerBroker2