PowerBroker2/SerialTransfer #18: calcOverhead Function Incorrect?

gfvalvo, 2020-04-15:
It looks to me like the calcOverhead function actually finds the **LAST**  occurrence of START_BYTE in the payload, not the first one as stated in the comments (and assumed by the unpackPacket function).

To find the **FIRST** occurrence, do:
```
void SerialTransfer::calcOverhead(uint8_t arr[], const uint8_t &len) {
  overheadByte = 0xFF;

  for (uint8_t i = 0; i < len; i++) {
    if (arr[i] == START_BYTE) {
      overheadByte = i;
      break;
    }
  }
}
```
Or, run the loop backwards.

PowerBroker2, 2020-04-15:
The documentation might be off. I originally tried a different way of doing COBS before settling on the design I have now. Let me do some testing to see if there is truly a bug.

PowerBroker2, 2020-04-15:
Yup, it's a bug. I'll fix it asap - thank you so much for bringing this to my attention!

PowerBroker2, 2020-04-15:
Should be fixed in [release 1.4.1](https://github.com/PowerBroker2/SerialTransfer/releases/tag/1.4.1)

closed 2020-04-15 by PowerBroker2