PowerBroker2/SerialTransfer #19: Missing transmission data

GranTesoro86, 2020-04-19:
**Describe the bug**
I tried to transfer the char array, but sometimes the data received by rx is missing

**To Reproduce**
Steps to reproduce the behavior:
Use these codes
RX 
```
#include "SerialTransfer.h"
boolean debugMode = true;
SerialTransfer DataTransfer;
struct STRUCT {
  char text[20];
} send_struct;
void setup() {
  Serial.begin(115200);
  Serial1.begin(115200);
  DataTransfer.begin(Serial1);
}
void loop( ) {
  uint16_t recSize = 0;
  if (DataTransfer.available()) {
    DataTransfer.rxObj(send_struct, sizeof(send_struct), recSize);
    recSize += sizeof(send_struct);
    if (debugMode) {
      Serial.println("DEBUG: Get data");
    }
    if (debugMode) {
      for (int i = 0; i < strlen(send_struct.text) ; i ++) {
        printf("%d\n", send_struct.text[i]);
      }
      Serial.println("");
    }
  }
  else if (DataTransfer.status < 0) {
    Serial.print("ERROR: ");
    Serial.println(DataTransfer.status);
  }
}
```
TX
```
#include "SerialTransfer.h"
#include <SoftwareSerial.h>
boolean debugMode = true;
SerialTransfer DataTransfer;
SoftwareSerial Serial2(D7, D8);
struct STRUCT {
  char text[20];
} send_struct;
void setup() {
  Serial.begin(115200);
  Serial2.begin(115200);
  DataTransfer.begin(Serial2);
}
void loop() {
  send_struct.text[0] = 168;
  send_struct.text[1] = 116;
  send_struct.text[2] = 192;
  send_struct.text[3] = 93;
  send_data();
}
void send_data() {
  uint16_t sendSize = 0;
  DataTransfer.txObj(send_struct, sizeof(send_struct), sendSize);
  sendSize += sizeof(send_struct);
  DataTransfer.sendData(sendSize);
  if (debugMode) {
    Serial.println("DEBUG: Send data");
  }
  delay(3000);
}

**Screenshots**
![image](https://user-images.githubusercontent.com/24543050/79684619-74bbe480-8265-11ea-8232-6b5399a77f16.png)
The first index of the char array received by rx is usually missing or incomplete

- SerialTransfer 1.4.1
- Board: NodeMCU and Arduino due
- Arduino IDE [1.8.12]

PowerBroker2, 2020-04-19:
2 things:

1.) NodeMCU has at least one extra hardware serial port. Why use SoftwareSerial?
2.) You don't have to send a struct if the struct only has one object inside of it. You can use `(class).txObj(array)` and `(class).rxObj(array)` to transmit a simple array.

Also, I was able to successfully validate the library using these sketches on two of my Arduino Megas:

TX:
```C++
#include "SerialTransfer.h"

SerialTransfer DataTransfer;

struct STRUCT {
  char text[20];
} send_struct;

boolean debugMode = true;

void setup()
{
  Serial.begin(115200);
  Serial1.begin(115200);
  DataTransfer.begin(Serial1);
}

void loop()
{
  send_struct.text[0] = 168;
  send_struct.text[1] = 116;
  send_struct.text[2] = 192;
  send_struct.text[3] = 93;
  
  send_data();
}

void send_data()
{
  DataTransfer.txObj(send_struct, sizeof(send_struct));
  DataTransfer.sendData(sizeof(send_struct));
  
  if (debugMode)
    Serial.println("DEBUG: Send data");
    
  delay(3000);
}

```

RX:
```C++
#include "SerialTransfer.h"

SerialTransfer DataTransfer;

struct STRUCT {
  char text[20];
} send_struct;

boolean debugMode = true;

void setup()
{
  Serial.begin(115200);
  Serial1.begin(115200);
  DataTransfer.begin(Serial1);
}

void loop( )
{
  if (DataTransfer.available())
  {
    DataTransfer.rxObj(send_struct, sizeof(send_struct));
    
    if (debugMode)
      Serial.println("DEBUG: Get data");
      
    if (debugMode)
    {
      for (uint16_t i=0; i<sizeof(send_struct.text); i++)
        if (send_struct.text[i] != 0)
          Serial.println((uint8_t)send_struct.text[i]);
    }
  }
  else if (DataTransfer.status < 0)
  {
    Serial.print("ERROR: ");
    Serial.println(DataTransfer.status);
  }
}

```

PowerBroker2, 2020-04-21:
No response yet, so I assume you have it all sorted out