PowerBroker2/SerialTransfer #20: Error sending data with variables

fatahillahardhi, 2020-04-23:
Hi, im newbie in Arduino, i try to sending multiple data with arduino uno and will received by nodemcu, in this case i will use variable data1 and data2, but nodemcu receive like this

[image]

this code will be implement to my project with 2 sensor, that's why i need variable for collect the data.

here's my arduino code (tx)
```
#include "SerialTransfer.h"
#include <SoftwareSerial.h>

SoftwareSerial mySerial(2, 3); // RX, TX
SerialTransfer myTransfer;
int data1 = 1;
int data2 = 2;

void setup()
{
  Serial.begin(115200);
  mySerial.begin(9600);
  myTransfer.begin(mySerial);
}

void loop()
{
  myTransfer.txBuff[0] = data1;
  myTransfer.txBuff[1] = data2;
  myTransfer.txBuff[2] = '\n';
  
  myTransfer.sendData(2);
  delay(100);
}
```

and my nodemcu code (rx)
```
#include "SerialTransfer.h"
#include <SoftwareSerial.h>

SoftwareSerial mySerial(D6, D5); // RX, TX
SerialTransfer myTransfer;

void setup()
{
  Serial.begin(115200);
  mySerial.begin(9600);
  myTransfer.begin(mySerial);
}

void loop()
{
  if(myTransfer.available())
  {
      Serial.print("Data 1 ");
      Serial.write(myTransfer.rxBuff[0]);
      Serial.println();
      Serial.print("Data 2 ");
      Serial.write(myTransfer.rxBuff[1]);
      Serial.println();
  }
  else if(myTransfer.status < 0)
  {
    Serial.print("ERROR: ");

    if(myTransfer.status == -1)
      Serial.println(F("CRC_ERROR"));
    else if(myTransfer.status == -2)
      Serial.println(F("PAYLOAD_ERROR"));
    else if(myTransfer.status == -3)
      Serial.println(F("STOP_BYTE_ERROR"));
  }
}
```

bytheway sorry for my english  :D

PowerBroker2, 2020-04-23:
Instead of stuffing the array `myTransfer.txBuff` manually (as your doing in the TX sketch), you should use `myTransfer.txObj()` and `myTransfer.rxObj()` to send and receive data. For example, try this:

TX:
```C++
#include "SerialTransfer.h"
#include <SoftwareSerial.h>

SoftwareSerial mySerial(2, 3); // RX, TX
SerialTransfer myTransfer;
int data1 = 1;
int data2 = 2;

void setup()
{
  Serial.begin(115200);
  mySerial.begin(9600);
  myTransfer.begin(mySerial);
}

void loop()
{
  uint8_t sendSize = 0;

  myTransfer.txObj(data1, sizeof(data1), sendSize);
  sendSize += sizeof(data1);

  myTransfer.txObj(data2, sizeof(data2), sendSize);
  sendSize += sizeof(data2);
  
  myTransfer.sendData(sendSize);
  delay(100);
}
```

RX:
```C++
#include "SerialTransfer.h"
#include <SoftwareSerial.h>

SoftwareSerial mySerial(D6, D5); // RX, TX
SerialTransfer myTransfer;
int data1;
int data2;

void setup()
{
  Serial.begin(115200);
  mySerial.begin(9600);
  myTransfer.begin(mySerial);
}

void loop()
{
  if (myTransfer.available())
  {
    uint8_t recSize = 0;

    myTransfer.txObj(data1, sizeof(data1), recSize);
    recSize += sizeof(data1);

    myTransfer.txObj(data2, sizeof(data2), recSize);
    recSize += sizeof(data2);

    Serial.print("Data 1 ");
    Serial.write(data1);
    Serial.println();
    Serial.print("Data 2 ");
    Serial.write(data2);
    Serial.println();
  }
  else if (myTransfer.status < 0)
  {
    Serial.print("ERROR: ");

    if (myTransfer.status == -1)
      Serial.println(F("CRC_ERROR"));
    else if (myTransfer.status == -2)
      Serial.println(F("PAYLOAD_ERROR"));
    else if (myTransfer.status == -3)
      Serial.println(F("STOP_BYTE_ERROR"));
  }
}
```

fatahillahardhi, 2020-04-23:
Hi, @PowerBroker2 i've been trying ur code, and its not problem, but when i use this code to my project which is use 2 rfid readers (MFRC522), the nodemcu didnt receive the data, the result like this

COM5 is RX and COM3 is TX
[image]

and for now i use Arduino Mega 2560

code **Arduino Mega (TX)**
```
/**
 * --------------------------------------------------------------------------------------------------------------------
 * Example sketch/program showing how to read data from more than one PICC to serial.
 * --------------------------------------------------------------------------------------------------------------------
 * This is a MFRC522 library example; for further details and other examples see: https://github.com/miguelbalboa/rfid
 *
 * Example sketch/program showing how to read data from more than one PICC (that is: a RFID Tag or Card) using a
 * MFRC522 based RFID Reader on the Arduino SPI interface.
 *
 * Warning: This may not work! Multiple devices at one SPI are difficult and cause many trouble!! Engineering skill
 *          and knowledge are required!
 *
 * @license Released into the public domain.
 *
 * Typical pin layout used:
 * -----------------------------------------------------------------------------------------
 *             MFRC522      Arduino       Arduino   Arduino    Arduino          Arduino
 *             Reader/PCD   Uno/101       Mega      Nano v3    Leonardo/Micro   Pro Micro
 * Signal      Pin          Pin           Pin       Pin        Pin              Pin
 * -----------------------------------------------------------------------------------------
 * RST/Reset   RST          9             5         D9         RESET/ICSP-5     RST
 * SPI SS 1    SDA(SS)      ** custom, take a unused pin, only HIGH/LOW required **
 * SPI SS 2    SDA(SS)      ** custom, take a unused pin, only HIGH/LOW required **
 * SPI MOSI    MOSI         11 / ICSP-4   51        D11        ICSP-4           16
 * SPI MISO    MISO         12 / ICSP-1   50        D12        ICSP-1           14
 * SPI SCK     SCK          13 / ICSP-3   52        D13        ICSP-3           15
 *
 */
#include <Wire.h>
#include <SPI.h>
#include <MFRC522.h>
#include <SoftwareSerial.h>
#include "SerialTransfer.h"

#define RST_PIN         8          // Configurable, see typical pin layout above
#define SS_1_PIN        9         // Configurable, take a unused pin, only HIGH/LOW required, must be diffrent to SS 2
#define SS_2_PIN        7          // Configurable, take a unused pin, only HIGH/LOW required, must be diffrent to SS 1

#define NR_OF_READERS   2
SerialTransfer myTransfer;

byte ssPins[] = {SS_1_PIN, SS_2_PIN};

MFRC522 mfrc522[NR_OF_READERS];   // Create MFRC522 instance.
int i = 1;
byte code[10];
String uidString;
int count0 = 0;
int count1 = 0;

/**
 * Initialize.
 */
void setup() {
  Serial2.begin(9600);
  myTransfer.begin(Serial2);
  Serial.begin(115200); // Initialize serial communications with the PC
  while (!Serial);
  SPI.begin();        // Init SPI bus
  
  for (uint8_t reader = 0; reader < NR_OF_READERS; reader++) {
[... 129 lines cut ...]
    recSize += sizeof(data1);

    myTransfer.txObj(data2, sizeof(data2), recSize);
    recSize += sizeof(data2);

    Serial.print("Data 1 ");
    Serial.write(data1);
    Serial.println();
    Serial.print("Data 2 ");
    Serial.write(data2);
    Serial.println();
  }
  else if (myTransfer.status < 0)
  {
    Serial.print("ERROR: ");

    if (myTransfer.status == -1)
      Serial.println(F("CRC_ERROR"));
    else if (myTransfer.status == -2)
      Serial.println(F("PAYLOAD_ERROR"));
    else if (myTransfer.status == -3)
      Serial.println(F("STOP_BYTE_ERROR"));
  }
}
```

PowerBroker2, 2020-04-23:
```C++
  if (myTransfer.available())
  {
    uint8_t recSize = 0; // <---------- keep this here

    myTransfer.txObj(data1, sizeof(data1), recSize);
    recSize += sizeof(data1);

    myTransfer.txObj(data2, sizeof(data2), recSize);
    recSize += sizeof(data2);

    Serial.print("Data 1 ");
    Serial.write(data1);
    Serial.println();
    Serial.print("Data 2 ");
    Serial.write(data2);
    Serial.println();
  }
```

fatahillahardhi, 2020-04-23:
uh sorry! i was wrong to paste the code

this is **NodeMCU (RX)** code that i use and the result still same

```
#include <SoftwareSerial.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <WiFiClient.h>
#include "SerialTransfer.h"

const char* ssid = "******";
const char* password = "******";
const char* host = "******";
const int httpPort = 80;
WiFiClient client;

SoftwareSerial mySerial(D2, D1);
SerialTransfer myTransfer;
int count0;
int count1;

String request_string;
HTTPClient http;

void setup() {
  mySerial.begin(9600);
  Serial.begin(115200);
  myTransfer.begin(mySerial);
//  Serial.print("Connecting to ");
//  Serial.println(ssid);
//  WiFi.begin(ssid, password);
//  while (WiFi.status() != WL_CONNECTED)
//  {
//    delay(500);
//    Serial.print(".");
//  }
//  Serial.println();
//  Serial.println("Wifi connected");
//  Serial.println("IP adress: ");
//  Serial.println(WiFi.localIP());
//  Serial.println("NesMask: ");
//  Serial.println(WiFi.subnetMask());
//  Serial.println("Getway: ");
//  Serial.println(WiFi.gatewayIP());
  //End

}

void loop() {

  if (myTransfer.available() ) {
    uint8_t recSize = 0;
    myTransfer.txObj(count0, sizeof(count0), recSize);
    recSize += sizeof(count0);
    myTransfer.txObj(count1, sizeof(count1), recSize);
    recSize += sizeof(count1);
    Serial.println();
    Serial.print("Counter 0 = ");
    Serial.write(count0);
    Serial.println();
    Serial.print("Counter 1 = ");
    Serial.write(count1);
    Serial.println();
    delay(1000);
  }else if (myTransfer.status < 0){
    Serial.print("ERROR: ");
    if (myTransfer.status == -1)
      Serial.println(F("CRC_ERROR"));
    else if (myTransfer.status == -2)
      Serial.println(F("PAYLOAD_ERROR"));
    else if (myTransfer.status == -3)
      Serial.println(F("STOP_BYTE_ERROR"));
  }
}
```

PowerBroker2, 2020-04-23:
Try using `Serial.print(count#);` instead of `Serial.write(count#);`

fatahillahardhi, 2020-04-23:
the result is not same with the data, count0 and count1 only showing **0** everytime when i send the data

PowerBroker2, 2020-04-23:
Try these sketches:

TX:
```C++
/**
   --------------------------------------------------------------------------------------------------------------------
   Example sketch/program showing how to read data from more than one PICC to serial.
   --------------------------------------------------------------------------------------------------------------------
   This is a MFRC522 library example; for further details and other examples see: https://github.com/miguelbalboa/rfid

   Example sketch/program showing how to read data from more than one PICC (that is: a RFID Tag or Card) using a
   MFRC522 based RFID Reader on the Arduino SPI interface.

   Warning: This may not work! Multiple devices at one SPI are difficult and cause many trouble!! Engineering skill
            and knowledge are required!

   @license Released into the public domain.

   Typical pin layout used:
   -----------------------------------------------------------------------------------------
               MFRC522      Arduino       Arduino   Arduino    Arduino          Arduino
               Reader/PCD   Uno/101       Mega      Nano v3    Leonardo/Micro   Pro Micro
   Signal      Pin          Pin           Pin       Pin        Pin              Pin
   -----------------------------------------------------------------------------------------
   RST/Reset   RST          9             5         D9         RESET/ICSP-5     RST
   SPI SS 1    SDA(SS)      ** custom, take a unused pin, only HIGH/LOW required *
   SPI SS 2    SDA(SS)      ** custom, take a unused pin, only HIGH/LOW required *
   SPI MOSI    MOSI         11 / ICSP-4   51        D11        ICSP-4           16
   SPI MISO    MISO         12 / ICSP-1   50        D12        ICSP-1           14
   SPI SCK     SCK          13 / ICSP-3   52        D13        ICSP-3           15

*/
#include <Wire.h>
#include <SPI.h>
#include <MFRC522.h>
#include <SoftwareSerial.h>
#include "SerialTransfer.h"

#define RST_PIN         8          // Configurable, see typical pin layout above
#define SS_1_PIN        9         // Configurable, take a unused pin, only HIGH/LOW required, must be diffrent to SS 2
#define SS_2_PIN        7          // Configurable, take a unused pin, only HIGH/LOW required, must be diffrent to SS 1

#define NR_OF_READERS   2
SerialTransfer myTransfer;

byte ssPins[] = {SS_1_PIN, SS_2_PIN};

MFRC522 mfrc522[NR_OF_READERS];   // Create MFRC522 instance.
int i = 1;
byte code[10];
String uidString;
struct STRUCT{
  int count0;
  int count1;
} data;

/**
   Initialize.
*/
void setup()
{
  Serial2.begin(9600);
  myTransfer.begin(Serial2);
  Serial.begin(115200); // Initialize serial communications with the PC
  while (!Serial);
  SPI.begin();        // Init SPI bus

  for (uint8_t reader = 0; reader < NR_OF_READERS; reader++)
  {
    mfrc522[reader].PCD_Init(ssPins[reader], RST_PIN); // Init each MFRC522 card
[... 148 lines cut ...]
{
  if (myTransfer.available())
  {
    myTransfer.rxObj(data, sizeof(data));
    
    Serial.println();
    Serial.print("Counter 0 = ");
    Serial.write(data.count0);
    Serial.println();
    Serial.print("Counter 1 = ");
    Serial.write(data.count1);
    Serial.println();
  }
  else if (myTransfer.status < 0)
  {
    Serial.print("ERROR: ");
    if (myTransfer.status == -1)
      Serial.println(F("CRC_ERROR"));
    else if (myTransfer.status == -2)
      Serial.println(F("PAYLOAD_ERROR"));
    else if (myTransfer.status == -3)
      Serial.println(F("STOP_BYTE_ERROR"));
  }
}
```

fatahillahardhi, 2020-04-24:
Hi @PowerBroker2 , i've been trying this code, everything is ok for count0 there is no problem, but the error comes for count1, the data give random number for count0 even though i send data for count1.

[image]

fatahillahardhi, 2020-04-24:
hi @PowerBroker2 the problem has solved now, im trying to remove struct and use array, its work correctly, thank you very much for your help!!

closed 2020-04-24 by fatahillahardhi