PowerBroker2/SerialTransfer #23: Mirror library for computers/linux

BrainStone, 2020-06-29:
**Is your feature request related to a problem? Please describe.**
I'd like to be able to use this library to communicate with an arduino from a linux machine. However the project I'd like to implement this in cannot make use of the python mirror library.

**Describe the solution you'd like**
I'm looking for a computer/linux mirror library.

**Describe alternatives you've considered**
I've taken a look at using the python library but that's just not working out for me. If worst comes to worst I could imagine trying to port the library myself, but I think it would be to the benefit of everyone to have a C++ mirror library.

PowerBroker2, 2020-06-29:
When you say the Python library isn't working out for you, do you mean there's a bug in the Python library or your PC code isn't originally meant for Python?

BrainStone, 2020-06-29:
I mean that the project itself is written in C++ and I can't integrate a python library

PowerBroker2, 2020-06-29:
Ah, ok. I can port the lib to a Linux C++ library, but no promises on a date to have it complete by.

BrainStone, 2020-06-29:
That would be much apprechiated!

Do you have a rough guess on how long it'll take? I do understand not wanting to commit to a date and I won't bug you about it, but I'd love to know if it's worth waiting of if I need to look for other ways to get the linux computer talking.

PowerBroker2, 2020-06-29:
I can make an empty repo for it today. That way anyone in the community can contribute to it and progress will be faster. Also, I don't have a Linux machine or much experience with Linux, so I would need help with testing anyway.

Basically, I can start today, but might take a month or so to finish it. If others in the open source community help, it should go faster.

BrainStone, 2020-06-29:
Alright. I should be able to help out.

PowerBroker2, 2020-06-29:
Here's the link: [https://github.com/PowerBroker2/CSerialTransfer](https://github.com/PowerBroker2/CSerialTransfer)

BrainStone, 2020-06-29:
For testing you can just use WSL. Or at least for compiling.  
A virtual machine should work just fine too.

BrainStone, 2020-07-07:
Would you mind giving this project a start? Like with the build tools and the like. There's quite a few and I don't want to make that decision for you.

If you're asking me, I'd say CMake. Decent learning curve, works well with VisualStudio (VisualStudio has support for Linux CMake projects) and is overall commonly used.

Yash1903, 2021-01-27:
Hello @PowerBroker2  and @BrainStone can i send file using this library.(File already Exist) ?

PowerBroker2, 2021-01-27:
@Yash1903, yes, you can send a file with this library. I've seen someone transfer 2K jpg pictures from one Arduino to another with SerialTransfer. Check out the [example sketches](https://github.com/PowerBroker2/SerialTransfer/tree/master/examples) whos names are appended by `_file`

Wisaya, 2022-03-21:
Excuse me, but any news about this?

v-r-a, 2023-04-03:
+1 request!

Satnet, 2023-10-29:
That CSerialTransfer, I would very much appreciate if it became a reality. I wanna transfer data reliably between a Windows program and Arduino. Using SerialTransfer and CSerialTransfer seems like the perfect solution!

frozer, 2025-03-31:
@Satnet @shyney7 @BrainStone hey, I ported it to C#, check it out:
* https://github.com/frozer/SerialTransfer.NET
* https://www.nuget.org/packages/SerialTransfer.NET/

closed 2026-11-06 by PowerBroker2