PowerBroker2/SerialTransfer #24: Is it possible to add the ability to support RS485 Half Duplex?

dalise, 2020-06-29:
**Is your feature request related to a problem? Please describe.**
Is it possible to adapt the SerialTransfer library to work with a half duplex RS485 network?
I have two networks (one which I currently use the SerialTransfer library for), the 2nd network however is RS485 and hence I am looking at a library with the capabilities of the full duplex SerialTransfer library in a half duplex package.

**Describe the solution you'd like**
Ability to control the timing and TX/RX enable lines of an RS485 Receiver 
Management of transmitting and receiving structured data (ints, floats etc) in a polled network 
i.e. looking at the RS485 network having one master and the rest being slaves as opposed to a multi-master network using collision detection

**Describe alternatives you've considered**
Currently I am looking to manually control the enable TXE/RXE inputs, timing management and node addressing, however thought I would ask the question before going down the rabbit hole!

PowerBroker2, 2020-07-03:
Node addressing can be taken care of via the new field "packet ID" - basically messages from each node could have a unique "packet ID" value, thus acting as a sort of "node ID".

The flow control portion of your question is a little harder to solve. The library doesn't natively support software flow control logic, but you could manually implement your own software flow control logic at the sketch level. You'll have to get creative, but it's definitely possible.

Does that help?

dalise, 2020-07-03:
Its does indeed help! 
I have just been testing the packet ID feature and that has saved me a lot of work!
Also I agree, it makes sense to manage the flow control outside of the library as the baud rate will affect the critical timing required for the enable lines. Time for me to drag out the logic analyser :)
By the way, great work on the library, it's well documented, rock solid and easy to implement.

closed 2020-07-03 by dalise