PowerBroker2/SerialTransfer #27: Licensing

BrainStone, 2020-07-03:
Hey, I'd like to use this library and its siblings in a proprietary commercial product. Which is not possible with the current GPLv3 license.  
I'd like to discuss this matter with you in private but I sadly was not able to find contact information for contacting you in private.

I'd prefer you reach out to me over Discord: [image]
If you don't use Discord, you can contact me on my email: [image]
*(Note both contact infos are posted here as images to prevent crawlers from seeing them.)*

It would be much appreachiated if you delete the issue after you read it or at least got rid of the Discord username and the email.

Thank you for your work and time!

BrainStone, 2020-07-06:
Hi. I haven't heard from you on any of those channels. Would you mind reaching out? We intent to pay you for those special usage rights.

PowerBroker2, 2020-07-06:
Sorry, been thinking things over. I want everyone to be able to use my code for anything they want (as long as it isn't nefarious) whether it's for a closed or open source project. I'm not sure how the GPLv3 license prevents you from using it in a proprietary commercial product, but I went ahead and changed it to an MIT License anyway. I don't think the MIT one has any restrictions on use-cases.

Either way, I really appreciate your interest in the lib, but I'm not interested in payment - it makes life a lot simpler for me.

BrainStone, 2020-07-06:
Well the issue with the GPL licenses is that they require projects that use projects licensed under GPL to be licensed and published (both source and binary) under GPL too.  
In short it's a "contagious" license. And therefore very popular in OS projects as it makes sure that everything using those projects remains OS too. Though naturally there's also some controversy about that.

And yes an MIT license has no such restrictions.

And just to be clear we wanted to pay you for the work already done and were not expecting any kind of additonal or proritized support or feature implementations.

Also when changing license you have to ask all contributors if they are ok with the change away from GPL. While I don't mind either way it's something to keep in mind for the future. While in most cases all that stuff doesn't matter, the international trade laws surrunding licenses are very clear and courts have been enforcing them too (assuming someone takes it to court).  
Now I don't want to scare you will all that, it's just a bit of information.

Thank you so much for changing the license too!

closed 2020-07-06 by BrainStone