PowerBroker2/SerialTransfer #33: SerialTransfer::available() only reads 1 byte per call

BrainStone, 2020-07-21:
Alright. Either there's something wrong with your code or I'm going crazy.
But as I read the code calling `SerialTransfer::available()` only ever reads a single byte instead of trying to fetch the whole packet.

https://github.com/PowerBroker2/SerialTransfer/blob/234c7e35fd03cff5838c474b24b6247ab37eb165/src/SerialTransfer.cpp#L94-L109

Now if I am reading that correctly that is in fact an issue. Mainly as my current code only calls that function sparringly (once every 100ms) as there's lots of other stuff to do. Also the description of the method is misleading as that sounds like calling this method takes care of the entirety of the parsing.

Now I also think your intentions were to have this method as a byte by byte method and the prorgammer can determine the state of the parser by reading the `state` variable.  
Not only would this be a prime candidate for an enum but also it never gets set back to the state ``NO_DATA``.

Now I'd advise you to have the ``available`` function in all 3 interfaces parse either the entire serial data or until it reaches the end of a packet. Then additionally have a byte by byte method for those that want to work that way. 

And lastly I really liked the function return an error value instead of setting an internal state on error.

PowerBroker2, 2020-07-21:
Ah, good catch - another set of my airhead mistakes lol. I'll take a deeper look into it and fix it soon. Thanks for the heads up!

Also, with the Linux thing - I don't think I have enough time/experience with the OS to figure it out anytime soon.

PowerBroker2, 2020-07-21:
Should be fixed in release 3.0.2.

closed 2020-07-21 by PowerBroker2

BrainStone, 2020-07-21:
I think it's only byte by byte in SPI too. Forgot to mention that.

PowerBroker2, 2020-07-21:
I'm not very experienced with SPI so I could be wrong, but I was under the impression the ISR fires on each byte received automatically - there is no SPI.available(), is there?

BrainStone, 2020-07-21:
I'm not experienced either. Just something I noticed.