PowerBroker2/SerialTransfer #34: Error compiling examples on Arduino Nano 33 BLE

luke-iqt, 2020-07-21:
**Describe the bug**
The library will not compile on the Arduino Nano 33 BLE. I tried to compile the uart_tx_data example and got the following error:
````

/Users/lberndt/Documents/Arduino/libraries/SerialTransfer/src/SPITransfer.cpp: In member function 'uint8_t SPITransfer::available()':
/Users/lberndt/Documents/Arduino/libraries/SerialTransfer/src/SPITransfer.cpp:115:29: error: 'SPDR' was not declared in this scope
  volatile uint8_t recChar = SPDR;
                             ^~~~
/Users/lberndt/Documents/Arduino/libraries/SerialTransfer/src/SPITransfer.cpp:115:29: note: suggested alternative: 'SPI'
  volatile uint8_t recChar = SPDR;
                             ^~~~
                             SPI
exit status 1
Error compiling for board Arduino Nano 33 BLE.
````
I think that SPI maybe implemented differently on this board.

**To Reproduce**
Steps to reproduce the behavior:
1. Goto examples, board uart_tx_data
2. Select the Arduino Nano 33 BLE as Board Type
3. compile

**Desktop (please complete the following information):**
 - OS: MacOS 1.15.4
 - Arduino: 1.8.13

PowerBroker2, 2020-07-21:
The Arduino Nano 33 BLE is always a troublemaker...It might take a while to figure out how to interface with SPI correctly on an Arduino Nano 33 BLE due to lack of examples from a quick google search. However, if you know the SPI register names are for your Nano 33, let me know and I'll try and fix it.

finostro-amtc, 2020-07-22:
This bug also apllies to arduino due.

finostro-amtc, 2020-07-22:
I had to delete SPITransfer.h and cpp since I am not using spi. It would be nice to have a fix though.

PowerBroker2, 2020-07-22:
Should be fixed in [3.0.3](https://github.com/PowerBroker2/SerialTransfer/releases/tag/3.0.3). Unfortunately, since there wasn't a good fix for either board, I've decided to not support these boards in terms of SPI - UART and I2C functionality should still work, however.

closed 2020-07-22 by PowerBroker2

nup002, 2020-10-18:
This still does not work, at least on the Nano Every. I have tested both Version 3.0.3 and 3.0.4, both installed from the IDE Library manager.
Attempting to compile the example uart_tx_data leads to the following compiler error. I have replaced the arduino install directory with `<arduino dir>`.
```
<arduino dir>\libraries\SerialTransfer\src\SPITransfer.cpp: In member function 'uint8_t SPITransfer::available()':
<arduino dir>\libraries\SerialTransfer\src\SPITransfer.cpp:119:29: error: 'SPDR' was not declared in this scope
  volatile uint8_t recChar = SPDR;
                             ^~~~
<arduino dir>\libraries\SerialTransfer\src\SPITransfer.cpp:119:29: note: suggested alternative: 'SPI'
  volatile uint8_t recChar = SPDR;
                             ^~~~
                             SPI
exit status 1
Error compiling for board Arduino Nano Every.
```

It is frustrating and intuitive to me that trying to compile a project that does not use SPI results in errors related to SPI.

PowerBroker2, 2020-10-18:
You can delete the SPI header file to fix the issue. Currently working on a PR to finally resolve this issue [here](https://github.com/PowerBroker2/SerialTransfer/pull/47).

Sorry for the inconvenience - SPI is handled terribly by Arduino and makes it difficult to make libs for it.