PowerBroker2/SerialTransfer #38: No messages received after buffer overflow

finostro-amtc, 2020-07-27:
**Describe the bug**
I my code (with several message types), If for some reason the listening code stops calling available() for enough time to fill the buffer. After I get a CRC ERROR and then no messages are received. Sometimes I get no CRC ERROR but no messages are received. 

**To Reproduce**
Have not tried to reproduce With a simple example yet. 

**Expected behavior**
After an error one or two messages are lost but communication is recovered.

PowerBroker2, 2020-07-27:
Might be issues with delays in your code - can you post your entire TX and RX code?

finostro-amtc, 2020-07-28:
I am trying (unsuccessfully) to reproduce the error with a smaller example. 
The code is a bit long, If I cannot reproduce the error I will post the communication part.

PowerBroker2, 2020-08-05:
Can't reproduce issue - must be user error (probably due to blocking code) - closing issue

closed 2020-08-05 by PowerBroker2