PowerBroker2/SerialTransfer #39: Issue with ESP8266

fernass, 2020-07-27:
Hi guys,
I'm using the SerialTransfer lib in order to send data from an ESP8266 (NodeMCU Lua Amica Modul V2 ESP8266 ESP-12F) to an Arduino Mega2560. Both sketches, for ESP8266 and Arduino are written and uploaded with the Arduino IDE (version 1.8.13).
Up to version 2.1.2 the attached code for the ESP8266 module compiled, linked and worked fine. After updating the library to 3.0.3 (I tried also all the versions in between) the code doesn't compile anymore and exits with the error message:

D:\Dokumente\Arduino\libraries\SerialTransfer\src\SPITransfer.cpp: In member function 'uint8_t SPITransfer::available()':
D:\Dokumente\Arduino\libraries\SerialTransfer\src\SPITransfer.cpp:119:29: error: 'SPDR' was not declared in this scope
  volatile uint8_t recChar = SPDR;

Is this issue known? Does the current version 3.0.0 and above not support the ESP8266 modules anymore?

By the way: version 3.0.3 works also with the Arduino.

Best regards
Fernass

[WiFiClient_OpenWeatherMap.zip](https://github.com/PowerBroker2/SerialTransfer/files/4984554/WiFiClient_OpenWeatherMap.zip)

PowerBroker2, 2020-07-27:
Yeah, with release 3.0.0 I added support for SPI and I2C communication. For some reason, though, not all Arduinos have the same SPI API and it breaks the entire library. I wasn't (and still aren't) sure which boards it is broken for.

You can delete SPITransfer.h and SPITransfer.cpp to fix it relatively easily yourself if you're cool with that.

I don't want to make any changes to the library right now because I'm currently coordinating [a fairly large update](https://github.com/PowerBroker2/SerialTransfer/pull/37) that should be implemented soon. I will, however, make sure the root cause will be fixed in that update.

closed 2020-08-02 by PowerBroker2

trungie, 2020-09-02:
I can confirm this is an issue on 3.0.3. Looking forward to #pr37.

PowerBroker2, 2020-09-02:
I'm currently working on a couple of other projects rn, but at some point the PR will go through