PowerBroker2/SerialTransfer #40: I use library it's really useful and solve my problem for variable communication over Arduino's which i tried to found over 2 years ago but it will be best library if you add function like Esp32 to Esp32 communication over WiFi or Bluetooth / Arduino to Arduino bidirectional communication over Bluetooth modules

kayyumuddin, 2020-07-30:


PowerBroker2, 2020-07-30:
I'm not sure I understand the problem, can you explain your issue? You can use this library with ESP32 modules to include the ESP32 built-in serial bluetooth already. Are you asking for an additional example or an actual new library function? Did you try to communicate between two ESP32s over bluetooth with the lib?

You shouldn't need a lib like this if you're using WiFi.

kayyumuddin, 2020-07-31:
Additional example,you didn't mention any serial bluetooth in Readme.md

PowerBroker2, 2020-07-31:
That comes down to being familiar with your ESP32 - check out [this](https://randomnerdtutorials.com/esp32-bluetooth-classic-arduino-ide/). You can integrate it with the SerialTransfer library by passing a `BluetoothSerial` class instance into `SerialTransfer.begin()` like you would any other `Serial` or `SoftwareSerial` class instance

closed 2020-07-31 by PowerBroker2