PowerBroker2/SerialTransfer #43: Stop the reading loop as soon as packet.status is not "Continue"

m-dhooge, 2020-09-14:
In SerialTransfer::available(), if port->available() is still _True_ when packet has decoded the EndOfFrame byte, the received frame is happily dumped to start receiving the new one.

To Reproduce:
Set the baudrate at 115200 and always send a second EndOfFrame (this was my idea to "ease recovery" in case a byte is not received…). Strangely, this is easier to reproduce with a real UART than with the SerialUSB emulation.

Expected behavior
Whenever a frame is received and ready, I should be able to process it ;-)

PowerBroker2, 2020-09-14:
Interesting - I'll have to take a look and see what's going on

PowerBroker2, 2020-09-14:
Ok, if you go into your local `SerialTransfer.cpp` file and find this function:

```C++
/*
 uint8_t SerialTransfer::available()
 Description:
 ------------
  * Parses incoming serial data, analyzes packet contents,
  and reports errors/successful packet reception
 Inputs:
 -------
  * void
 Return:
 -------
  * uint8_t bytesRead - Num bytes in RX buffer
*/
uint8_t SerialTransfer::available()
{
	bool    valid   = false;
	uint8_t recChar = 0xFF;

	if (port->available())
	{
		valid = true;

		while (port->available())
		{
			recChar = port->read();

			bytesRead = packet.parse(recChar, valid);
			status    = packet.status;
		}
	}
	else
	{
		bytesRead = packet.parse(recChar, valid);
		status    = packet.status;
	}

	return bytesRead;
}
```

Now, is the problem fixed if you replace the above function with this edited version?:

```C++
/*
 uint8_t SerialTransfer::available()
 Description:
 ------------
  * Parses incoming serial data, analyzes packet contents,
  and reports errors/successful packet reception
 Inputs:
 -------
  * void
 Return:
 -------
  * uint8_t bytesRead - Num bytes in RX buffer
*/
uint8_t SerialTransfer::available()
{
	bool    valid   = false;
	uint8_t recChar = 0xFF;

	if (port->available())
	{
		valid = true;

		while (port->available())
		{
			recChar = port->read();

			bytesRead = packet.parse(recChar, valid);
			status    = packet.status;

			if (status == NEW_DATA) // <<------ NEW ***
				break;                       // <<------ NEW ***
		}
	}
	else
	{
		bytesRead = packet.parse(recChar, valid);
		status    = packet.status;
	}

	return bytesRead;
}
```

m-dhooge, 2020-09-15:
Thanks for your quick answer :-)

What worked for me is:  if (status != CONTINUE)

This is more or less the same, but with CONTINUE, you get also the error statuses ;-)

closed 2020-09-15 by PowerBroker2