PowerBroker2/SerialTransfer #44: Error compiling on SAMD21

pegrei, 2020-09-30:
**Describe the bug**
I get following errors when including the `"SerialTransfer.h"` header in PlatformIO for a SAMD21:
```
 #include "Wire.h"
          ^~~~~~~~
compilation terminated.
.pio\libdeps\interferometer\SerialTransfer\src\SPITransfer.cpp: In member function 'uint8_t SPITransfer::available()':
.pio\libdeps\interferometer\SerialTransfer\src\SPITransfer.cpp:111:29: error: 'SPDR' was not declared in this scope
  volatile uint8_t recChar = SPDR;
                             ^~~~
.pio\libdeps\interferometer\SerialTransfer\src\SPITransfer.cpp:111:29: note: suggested alternative: 'SPI1'
  volatile uint8_t recChar = SPDR;
                             ^~~~
                             SPI1
*** [.pio\build\interferometer\libbc2\SerialTransfer\I2CTransfer.cpp.o] Error 1
*** [.pio\build\interferometer\libbc2\SerialTransfer\SPITransfer.cpp.o] Error 1
```
After deleting the SPI and I2C headers from the library the errors disappear.

**Desktop (please complete the following information):**
 - OS: Win10
 - PlatformIO with Arduino Framework and Atmel SAM Core

PowerBroker2, 2020-09-30:
Yes, unfortunately, the SPI API is different across Arduino architectures (and VERY annoyingly so), which causes some boards to fail to compile. As you've discovered, if you delete the SPI portion of the lib, everything else works fine. I'll have to add this on the list of unsupported MCUs in [the current dev version of the library](https://github.com/PowerBroker2/SerialTransfer/pull/37). Once the live library is updated, you won't be able to use the SPI portion of the library for your SAMD21, but you won't have a compile error either.

PowerBroker2, 2020-10-18:
Should be fixed with release [3.0.5](https://github.com/PowerBroker2/SerialTransfer/releases/tag/3.0.5)

closed 2020-10-18 by PowerBroker2