PowerBroker2/SerialTransfer #45: SerialTransfer for contemporary send and receive

ronart7329, 2020-10-06:
Good morning, first of all congratulations for the excellent library, but I'm having a problem when I try to do tx and rx from two nodes simultaneously, the nodes locks without apparently send and receive nothing.

Many thanks.

Arturo

PowerBroker2, 2020-10-06:
Can you post a schematic or clear description of your project? Also please post your entire code for both nodes.

ronart7329, 2020-10-06:
My goal is to have a sensor node that I can also configure remotely through a gw node on a serial connected to a kq300 powerline device.
the code:

[tx_sensore_node.txt](https://github.com/PowerBroker2/SerialTransfer/files/5337196/tx_sensore_node.txt)
[rx_gw_node.txt](https://github.com/PowerBroker2/SerialTransfer/files/5337202/rx_gw_node.txt)

the schema:

node sensor (tx) data (temperature, voltage, etc..) --> node gateway (rx) data
node gateway (tx) command (for configure, control, etc.) to ---> node sensor (tx)

That's all.

Many thanks.

Arturo

PowerBroker2, 2020-10-07:
TX:
```C++
#include "SerialTransfer.h"
#include <SoftwareSerial.h>
#include <PZEM004Tv30.h>
#include <Adafruit_Sensor.h>
#include <DHT.h>
#include <DHT_U.h>
#include <Wire.h>
#include "DS3231.h"

#define DHTPIN 4     // Digital pin connected to the DHT sensor 
// Feather HUZZAH ESP8266 note: use pins 3, 4, 5, 12, 13 or 14 --
// Pin 15 can work but DHT must be disconnected during program upload.

// Uncomment the type of sensor in use:
//#define DHTTYPE    DHT11     // DHT 11
#define DHTTYPE    DHT22     // DHT 22 (AM2302)
//#define DHTTYPE    DHT21     // DHT 21 (AM2301)

#define MAX_CMD  12
#define nodeId  1

SoftwareSerial SoftSerial(2, 3); // RX, TX

/* Use software serial for the PZEM
 * Pin 11 Rx (Connects to the Tx pin on the PZEM)
 * Pin 12 Tx (Connects to the Rx pin on the PZEM)
*/
PZEM004Tv30 pzem(&Serial3);

SerialTransfer myTransfer;

DHT_Unified dht(DHTPIN, DHTTYPE);

RTClib RTC;

DateTime now;

uint16_t sendSize = 0;

uint16_t recSize = 0;

uint32_t delayMS;

char buffer_in[32];

struct Data
{
    uint16_t id_node;
    uint16_t light = 0;
    float temperature = 0;
    float humidity = 0;
    float voltage = 0;
    float current = 0;
    float power = 0;
    float frequency = 0;
    unsigned long int time_ref = 0;
    
};

Data pkt;

struct Data readSensors(void)
{
  
  sensors_event_t event;
  dht.temperature().getEvent(&event);
  if (isnan(event.temperature)) {
    Serial.println(F("Error reading temperature!"));
[... 238 lines cut ...]
    Serial.print(";");
    Serial.print(pkt.time_ref);
    Serial.print(";");
    Serial.print(pkt.voltage);
    Serial.print(";");
    Serial.print(pkt.power);
    Serial.print(";");
    Serial.print(pkt.current);
    Serial.println();   
    if(recSize==32) {
      sendSize = 0;
      String msg = "Pkt rx=" + sendSize;
      msg.toCharArray(message_buffer,32);
      sendSize = myTransfer.txObj(message_buffer, sendSize);
      myTransfer.sendData(sendSize);
    }

    //recSize = myTransfer.rxObj(node_status, recSize);
    //Serial.println(node_status);
  }

  
  delay(1000);
}
```

PowerBroker2, 2020-10-07:
Softwareserial is only half duplex. In order to simultaneously send and receive, you must use a hardware UART port. You could also use 2 softwareserial ports on each Arduino - one for each communication direction.

closed 2020-10-07 by PowerBroker2