PowerBroker2/SerialTransfer #46: Library work with Arduino UNO and ESP32

blokerun, 2020-10-17:
I cannot transfer different types of data between Arduino Uno and ESP32. If the data is of only one type, for example "float", then everything works, but if I use different types in the structure, for example "float" and "char", then the data on the receiving side (ESP32) is distorted. How to make the library work with different controllers?

PowerBroker2, 2020-10-18:
Both boards are little-endian, so there shouldn't be a problem there (especially if sending just a float works). As for sending multiple values, it should work just fine and the problem is likely in the sketch. Can you post your code?

blokerun, 2020-10-23:
This code wrote to Arduino UNO- transmitter

`// This code for Arduino UNO
#include "SerialTransfer.h"
SerialTransfer myTransfer;

struct STRUCT {
  char z;
  float y;
} testStruct;

void setup()
{
  Serial.begin(115200);
  myTransfer.begin(Serial);
  testStruct.z = '$';
  testStruct.y = 4.5;
}

void loop()
{
  // use this variable to keep track of how many
  // bytes we're stuffing in the transmit buffer
  uint16_t sendSize = 0;
  ///////////////////////////////////////// Stuff buffer with struct
  sendSize = myTransfer.txObj(testStruct, sendSize);

  ///////////////////////////////////////// Send buffer
  myTransfer.sendData(sendSize);
  delay(500);
}`

This code wrote to ESP32 - receiver
`// This code for ESP32
#include "SerialTransfer.h"
SerialTransfer myTransfer;

struct STRUCT {
  char z;
  float y;
} testStruct;

void setup()
{
  Serial.begin(115200);
  Serial2.begin(115200);
  myTransfer.begin(Serial2);
}

void loop()
{
  if(myTransfer.available())
  {
    // use this variable to keep track of how many
    // bytes we've processed from the receive buffer
    uint16_t recSize = 0;

    recSize = myTransfer.rxObj(testStruct, recSize);
    Serial.print(testStruct.z);
    Serial.print(testStruct.y);
    Serial.print(" | ");

  }
}`

In the Serial port ESP32 printed:         $0.00 | $0.00 | $0.00 | $0.00 | $0.00 | $0.00 |

PowerBroker2, 2020-10-23:
Can you post your whole code - both sides?

blokerun, 2020-10-24:
I've simplified my code to make it clearer. If there is a boolean variable in the structure - the data at the reception is distorted, if I remove the boolean variable - the code works fine
```
// This code for Arduino UNO
#include "SerialTransfer.h"

SerialTransfer myTransfer;

struct STRUCT {
 
  int32_t z;
  float y;
  boolean flag1;//       without a boolean variable works, with this variable - no

} ;

STRUCT testStruct;

void setup()
{
  Serial.begin(115200);
  myTransfer.begin(Serial);
  
  testStruct.z = 10;
  testStruct.y = 0.2;
  testStruct.flag1 = false;
  
  
}

void loop()
{
  // use this variable to keep track of how many
  // bytes we're stuffing in the transmit buffer
  uint16_t sendSize = 0;

  ///////////////////////////////////////// Stuff buffer with struct
  sendSize = myTransfer.txObj(testStruct, sendSize);

  ///////////////////////////////////////// Send buffer
  myTransfer.sendData(sendSize);
  delay(500);
}
```

```
// This code for ESP32
#include "SerialTransfer.h"

SerialTransfer myTransfer;

struct STRUCT {
 
  int32_t z;
  float y;
  boolean flag1; //       without a boolean variable works, with this variable - no

} ;

STRUCT testStruct;

void setup()
{
  Serial.begin(115200);
  Serial2.begin(115200);
  myTransfer.begin(Serial2);
}

void loop()
{
  if(myTransfer.available())
  {
    // use this variable to keep track of how many
    // bytes we've processed from the receive buffer
    uint16_t recSize = 0;

    recSize = myTransfer.rxObj(testStruct, recSize);
	
    Serial.print(testStruct.z);
    Serial.print("_");
    Serial.print(testStruct.y);
    Serial.print("_");
    Serial.print(testStruct.flag1);
    Serial.print(" | ");

   
    Serial.println();

 
  }
}
```

akira-kun, 2020-11-11:
Does serialtransfer Uart, i2c ans SPI works with esp8266?

PowerBroker2, 2020-11-11:
It works with UART and I2C, but I don't think it's supported for SPI

akira-kun, 2020-11-12:
Very nice, and which one is faster and realible? I2c or serial uart? I need transfer large mp3 files downloaded from https

PowerBroker2, 2020-11-12:
Max clock rate for most arduino boards is 400KHz, but UART maxes out at a baud of 2 million, so UART would be faster. You should do a little research and see if the ESP8266 can do I2C at over 2MHz - that would be the deciding factor.

akira-kun, 2020-11-13:
I2C communication between two esp8266 could not be done, I've been trying to use wire.begin in many ports for SDA, SCL, but no way to make it go. Do you know what could be? Now Serial runs like a sharme.

PowerBroker2, 2020-12-27:
@blokerun 

My apologies for getting back to you so late, but I think I found the solution to your issue: When specifying your struct, use the "packed" attribute to make sure the struct is compatible across Arduinos. Here is how you would do it:

```C++
struct __attribute__((__packed__)) STRUCT {
  int32_t z;
  float y;
  boolean flag1; //       without a boolean variable works, with this variable - no
} ;
```

closed 2020-12-29 by PowerBroker2