PowerBroker2/SerialTransfer #49: uart read data example problem in ESP-01

alezulu73, 2020-11-10:
**Describe the bug**
I am trying to run the example uart_rx_data in a ESP-01 board. But when I verify the code I obtain this output:

C:\Users\Familia\Documents\Arduino\libraries\SerialTransfer\src\SPITransfer.cpp: In member function 'uint8_t SPITransfer::available()':
C:\Users\Familia\Documents\Arduino\libraries\SerialTransfer\src\SPITransfer.cpp:111:29: error: 'SPDR' was not declared in this scope
  volatile uint8_t recChar = SPDR;
                             ^
exit status 1
Error compiling for board Generic ESP8266 Module.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Arduino IDE: 1.8.13

Any help to solve this issue. 

Thank you

PowerBroker2, 2020-11-10:
Made some edits to the master branch, can you test with these changes and confirm the error is fixed?

PowerBroker2, 2020-11-10:
See latest release

closed 2020-11-10 by PowerBroker2