PowerBroker2/SerialTransfer #50: Sending list from arduino UNO to python using  SerialTransfer and PySerialTransfer

ericferreri, 2020-12-17:
Hello,
thanks for sharing your libs freely , 
I require some insights,
rather than a bug it is perhaps more lacking info to make run a very simple task,  here comes my issue when I try to send a small array with 2 integers in it form an arduibo to a pc:

on the arduino UNO side:
///////////////////////begining of arduino script////////////////
//////I would like to send a list 
#include "SerialTransfer.h"
SerialTransfer myTransfer;
void setup()
{
  Serial.begin(115200);
  myTransfer.begin(Serial);
}

int arry[]={1,4};
void loop()
{
     myTransfer.sendData(arry);
     delay(100);
}
////////////////////////////////end of arduino script////////////////////////////////////////////////////////

```
On the PC running Python:

######begining of Python script#############################
#### I would like to receive an array with 2 ints in it
import time
from pySerialTransfer import pySerialTransfer as txfer

if __name__ == '__main__':
    try:
        link = txfer.SerialTransfer('COM15')
        
        link.open()
        time.sleep(2) # allow some time for the Arduino to completely reset
        
        while True:
            send_size = 0
            
            ###################################################################
            # given just to define type and lenght of the expected data
            ###################################################################
            list_ = [1, 3]
            list_size = link.tx_obj(list_)
            send_size += list_size

         
            
            ###################################################################
            # Wait for a response and report any errors while receiving packets
            ###################################################################
            while not link.available():
                if link.status < 0:
                    if link.status == txfer.CRC_ERROR:
                        print('ERROR: CRC_ERROR')
                    elif link.status == txfer.PAYLOAD_ERROR:
                        print('ERROR: PAYLOAD_ERROR')
                    elif link.status == txfer.STOP_BYTE_ERROR:
                        print('ERROR: STOP_BYTE_ERROR')
                    else:
                        print('ERROR: {}'.format(link.status))
            
            ###################################################################
            # Parse response list
            ###################################################################
            rec_list_  = link.rx_obj(obj_type=type(list_),obj_byte_size=list_size,list_format='i')

            print('RCVD: {}'.format(rec_list_))
            print(' ')
    
    except KeyboardInterrupt:
        try:
            link.close()
        except:
            pass
    
    except:
        import traceback
        traceback.print_exc()
        
        try:
            link.close()
        except:
            pass

################end of python script##########################

I was not able to send a small array using your libs... here is the error I am getting in python:

Traceback (most recent call last):
  File "C:/Users/admin/Desktop/WPy32-3860/scripts/read_NANO_LIST.py", line 27, in <module>
    while not link.available():
  File "C:\Users\admin\Desktop\WPy32-3860\python-3.8.6\lib\site-packages\pySerialTransfer\pySerialTransfer.py", line 477, in available
    self.rxBuff[self.payIndex] = recChar
IndexError: list assignment index out of range
---------------------------

Thanks for helping me to solve this :)

PowerBroker2, 2020-12-18:
You need to use `int32_t` to specify the length of each int since Python defaults to 32-bit ints while most Arduinos default to 16-bits

ericferreri, 2020-12-18:
Thanks for the answer, in addition to that, SendDatum MUST be used instead of SendData, which might seem a bit counter-intuitive. Anyway, now everything works like a charm.
Thanks for the answer and the great libs :)

closed 2020-12-18 by CognitiveConnexionist

asafbokobza, 2020-12-30:
> You need to use `int32_t` to specify the length of each int since Python defaults to 32-bit ints while most Arduinos default to 16-bits

Hi
I have same problem. 
Could you please share with us an code examples for this.
Thank you and happy new year 
Asaf