PowerBroker2/SerialTransfer #51: How do we specify which "node" an I2C packet gets sent to?

chrisco484, 2021-01-23:
With serial there is only one other node - the one at the end of the serial link.

However with I2C working in a "multi master" environment, any node can send to any other node. 

How does the I2CTransfer get told the address of the node to send a frame to?

PowerBroker2, 2021-01-23:
If you take a look at `sendData()` for I2C, the last parameter `targetAddress` allows you to choose what I2C slave will receive the packet. See below:

https://github.com/PowerBroker2/SerialTransfer/blob/950e7520028557a2f3b6af00f2b2ad85a7f8ffa3/src/I2CTransfer.cpp#L47-L76

chrisco484, 2021-01-24:
> If you take a look at `sendData()` for I2C, the last parameter `targetAddress` allows you to choose what I2C slave will receive the packet. See below:

Doh! I must have missed that. That's awesome, thanks!

closed 2021-01-25 by PowerBroker2