PowerBroker2/SerialTransfer #52: Resend when any transfer error

luc-github, 2021-01-27:
Hi thank you for your work 😄 ,
 
I do not know if it is a feature request or documentation request

I see there are callback and error status  (CRC_ERROR/PAYLOAD_ERROR/STOP_BYTE_ERROR) but I do not understand if there is a way to ask for resend specific packet if any error happen. and eventually cancel transfer if error is not recoverable

Is this feature missing or I miss read the sources/examples ?

Thank you

PowerBroker2, 2021-01-27:
There isn't a way to automatically request a packet resend. If you need a resend, you'll have to figure out what the best way to request a packet resend for your specific project. It could be as simple as a digital IO signal or as complicated as sending a specific packet to the original sender to trigger a resend.

luc-github, 2021-01-27:
yes my guessing was it need to be handle using callback - right ?
but I do not see how the callback get the status error

PowerBroker2, 2021-01-27:
You don't need to use callbacks, but if you do use callbacks, you can still access the status member of the SerialTransfer class like this:

```C++
void someCallback()
{
  if (myTransfer.status <= 0)
  {
    // trigger resend
  }
}
```

luc-github, 2021-01-27:
excellent thank you ^_^

closed 2021-01-27 by luc-github