PowerBroker2/SerialTransfer #53: IC2 slave examples use reserved address 0

chrisco484, 2021-01-29:
I2C has 16 reserved addresses which nodes should avoid using as their addresses.

The first 8 (0-7) and last 8 addresses are reserved.

So:
`Wire.begin(0);`
is actually using the reserved address '0'.

The first "usable" address for IC2 nodes is 8 so maybe that would be a good choice for the sample.

Details of the reserved address space can be found here:
https://www.totalphase.com/support/articles/200349176-7-bit-8-bit-and-10-bit-I2C-Slave-Addressing#reserved

PowerBroker2, 2021-01-29:
I believe the examples were tested and validated as currently written and therefore I think address 0 is usable even if technically "not correct". If you feel it's that important, go ahead and make a PR (also note that the default argument mentioned in the previous issue should also be updated)

closed 2021-02-11 by PowerBroker2

chrisco484, 2021-02-11:
> I believe the examples were tested and validated as currently written and therefore I think address 0 is usable even if technically "not correct". If you feel it's that important, go ahead and make a PR (also note that the default argument mentioned in the previous issue should also be updated)

It will definitely work in most isolated testing environments. It's when it's added to an existing I2C "network" with a master that's doing broadcasts, and other "mastery" things where the wheels will fall off:

This forum topic discusses I2C broadcasts - which use the reserved adddress "0" (the forum is in relation to Pi but the broadcast concept is specified in the I2C standards so it applies to any I2C network implementation):

https://www.raspberrypi.org/forums/viewtopic.php?t=78585

chrisco484, 2021-02-11:
Reading further it actually could be dangerous sending a message to a slave at address 0.

I noticed this page in I2C org where is states that when sending to address 0 the following byte represents a command e.g. software reset. Every slave that listens for commands at the broadcast address will perform that command - yikes!

https://www.i2c-bus.org/addressing/general-call-address/