PowerBroker2/SerialTransfer #57: Reduce library memory usage

benknight135, 2021-02-24:
**Is your feature request related to a problem? Please describe.**
I am trying to use this library as part of a project with multiple devices connected with software serial. This libraries large size means I can't use 3 software serials on the Arduino Uno as this makes the memory approx. 110% dynamic memory just to declare all the SerialTransfer instances. 

**Describe the solution you'd like**
Reduce library memory usage.

**Describe alternatives you've considered**
I could use a Arduino Mega however I don't current have one available. 

**Additional context**
Here is the simplest code for this setup:
```
#include "SerialTransfer.h"
#include <SoftwareSerial.h>

SerialTransfer sTransfer_a;
SerialTransfer sTransfer_b;
SerialTransfer sTransfer_c;
SerialTransfer sTransfer_d;
SoftwareSerial sSerial_b(2,3);
SoftwareSerial sSerial_c(5,6);
SoftwareSerial sSerial_d(8,9);

void setup()
{
  Serial.begin(115200);
  sSerial_b.begin(115200);
  sSerial_c.begin(115200);
  sSerial_d.begin(115200);
  sTransfer_a.begin(Serial);
  sTransfer_b.begin(Serial);
  sTransfer_c.begin(Serial);
  sTransfer_d.begin(Serial);
}
void loop()
{
}
```

PowerBroker2, 2021-02-24:
Are you using all 4 SerialTransfer class instances to communicate over `Serial`, or did you mistype that?

Yeah, each class instance has 3 buffers of 255-ish bytes (one for receive, one for transmit, and one for CRC lookups). I understand there's a bit of overhead, but you can mitigate it to some degree if you do these things:

1. Minimize the number of ports used: Take advantage of the full duplex nature of hardware serial ports. If one device only transmits, another only receives, and both work on the same baud, connect both external devices to the same Arduino serial port. Similar things can be done with software serial ports if you can time the packets properly and use the `listen()` method.
2. Just don't use softwareserial (it's not a good lib to have to rely on)
3. Invest in an MCU that has more memory, more hardware ports, and faster. The uno is pretty low end. The best bang for your buck is a Teensy 3.5, but the Teensy 4.1 (the board I usually use) is one of the best Arduino compatible boards out there
4. If you know the max packet payload length you will ever receive or transmit, you can lower this number to decrease the size of each class instances' receive and transmit buffers:
https://github.com/PowerBroker2/SerialTransfer/blob/ee943104e14cad8a37d7b9b0118c1b45b8a308b4/src/Packet.h#L34

There are a couple of other things that could make it a little less memory intensive, but would require some rework of the lib.

The most important thing, however, is making sure your whole design is well planned to keep from running into memory problems.

benknight135, 2021-02-24:
Yes those extra serials where a typo. I've edited it to reflect how I intended. 

Thank you for the advice I will try out your suggestions. I need to communicate both ways on all the devices so it's quite complex to manage.

closed 2021-04-04 by PowerBroker2