PowerBroker2/SerialTransfer #58: Issue With UART Send/Receive

michaelwro, 2021-03-08:
Hi, I really like your SerialTransfer library. I think it would be a great product to integrate into my [quadcopter flight controller](https://github.com/michaelwro/HummingbirdFCU).

I am using your library to communicate between a Teensy 4.1 and Arduino Nano via [XBee3](https://www.sparkfun.com/products/15130) radios. I am following your `uart_rx_datum.ino` and `uart_tx_datum.ino` examples to send a struct from the Teensy to the Nano. I have debug messages enabled and am frequently getting `ERROR: CRC_ERROR` and `ERROR: STOP_BYTE_ERROR` messages at the receiver (Nano). These errors are significantly slowing down my data rate, since I assume you discard a packet if an error is encountered. I am not very familiar with the inner workings of your library, and thought I'd bring this issue to your attention. I wonder if I am using it improperly, or if there is a potential issue buried within the code.

Thank you for your time addressing this issue and for creating this great project!

## Teensy 4.1 (Transmitter)
```cpp
#include "SerialTransfer.h"
#define XBEE_UART Serial8
SerialTransfer XbeeTelem;

typedef struct
{
    float mx;
    float my;
    float mz;
} MagnetometerData_s;
MagnetometerData_s DataPacket {1.0f, 2.0f, 3.0f};

void setup()
{
    Serial.begin(115200);
    XBEE_UART.begin(115200);
    XbeeTelem.begin(XBEE_UART, true, Serial);  // send, debug, debug interface
    delay(200);
}

void loop()
{
    XbeeTelem.sendDatum(DataPacket, sizeof(DataPacket));
    delay(50);
}
```

## Arduino Nano (Receiver)
```cpp
#include <Arduino.h>
#include <SoftwareSerial.h>
#include "SerialTransfer.h"

SoftwareSerial XbeeSerial(2, 3);
SerialTransfer TelemStream;
void PrintData();

typedef struct
{
  float mx;
  float my;
  float mz;
} MagnetometerData_s;
MagData_s RecPacket;

void setup() {
  XbeeSerial.begin(115200);
  Serial.begin(115200);
  TelemStream.begin(XbeeSerial, true, Serial);
delay(200);
}

void loop() {
  if (TelemStream.available()) {
    TelemStream.rxObj(RecPacket);
    PrintData();
  }
}

void PrintData() {
  Serial.print("mx: "); Serial.print(RecPacket.mx, 2);
  Serial.print("  my: "); Serial.print(RecPacket.my, 2);
  Serial.print("  mz: "); Serial.println(RecPacket.mz, 2);
}
```

### Serial Output (Nano)
```
mx: 1.00  my: 2.00  mz: 3.00
ERROR: CRC_ERROR
mx: 1.00  my: 2.00  mz: 3.00
ERROR: CRC_ERROR
mx: 1.00  my: 2.00  mz: 3.00
mx: 1.00  my: 2.00  mz: 3.00
mx: 1.00  my: 2.00  mz: 3.00
ERROR: STOP_BYTE_ERROR
ERROR: CRC_ERROR
mx: 1.00  my: 2.00  mz: 3.00
mx: 1.00  my: 2.00  mz: 3.00
mx: 1.00  my: 2.00  mz: 3.00
mx: 1.00  my: 2.00  mz: 3.00
mx: 1.00  my: 2.00  mz: 3.00
```

PowerBroker2, 2021-03-08:
Thanks for your interest in my library and your project looks really cool.

The one thing that stands out to me as a potential problem is that you're using SoftwareSerial.h at 115200 baud. [SoftwareSerial is slow.](https://forum.arduino.cc/index.php?topic=339644.0) You would probably not have any issues with the sketches if you either used the one hardware serial port on the Uno, used AltSoftwareSerial, or used another MCU that has an extra hardware serial port available.

You could try hardwiring the TX/RX/GND pins between the two MCUs and test without the XBees themselves. This would simplify the debugging.

Seems you're more interested in quadcopters, but you might be interested in taking a look at one of my other libraries - [one meant for fixed wing Arduino RC planes](https://github.com/PowerBroker2/ArdUAV).

PowerBroker2, 2021-03-08:
Side note: you don't need to specify the second parameter in `sendDatum()`, you can just do the following:

```C++
XbeeTelem.sendDatum(DataPacket);
```

michaelwro, 2021-03-08:
Thanks for the quick reply!

The reason I am using SoftwareSerial on the Nano is because I designed a small PCB for it and arbitrarily chose those pins for TX/RX. I've run into other issues with the PCB/Nano and am considering redesigning and upgrading to something like a Teensy 3.x (with more _hardware_ serial ports, too). I will investigate using a slower baud rate and see if that solves it.

I've come across your work in the past and found it very useful. Thanks again!

[image]

closed 2021-03-11 by PowerBroker2