PowerBroker2/SerialTransfer #59: Confused on how to uncomment SPITransfer.h

K-Space909, 2021-03-16:
I'm not sure why SPITransfer.h comes commented out, but uncommenting it is causing issues. I have no idea what parts are needed, and some of the variables are undefined. There seems to be a missing semicolon or curly brackets with contents.

If it matters, I'm compiling for a SAMD21 cortex M0+ board (the Seeeduino Xiao). I'll attach a screenshot of the way I've got it commented out at the moment. Please note that I'm still a noob top coding and this is the first time I've tried this library, so it's very likely that I'm just not seeing how it's intended to work. 

[image]

I'm getting:

    SPITransfer.h: 26:36: error: 'T' does not name a type; did you mean 'Tc'?
    uint16_t SPITransfer*: txObj(const T &val, const uint16_t &index=0, const uint16_t &len=sizeof(T))

    SPITransfer.h: 26:98: error: expected ';' at end of member declaration
    uint16_t SPITransfer*: txObj(const T &val, const uint16_t &index=0, const uint16_t &len=sizeof(T))

    SPITransfer.h: 26:96: error: 'T' was not declared in this scope
    uint16_t SPITransfer*: txObj(const T &val, const uint16_t &index=0, const uint16_t &len=sizeof(T))
    SPITransfer.h:26: note  suggested alternative  Tc
    uint16_t SPITransfer*: txObj(const T &val, const uint16_t &index=0, const uint16_t &len=sizeof(T))

PowerBroker2, 2021-03-16:
> I'm not sure why SPITransfer.h comes commented out, but uncommenting it is causing issues. 

SPITransfer.h comes commented out because of all the issues it caused ([like this](https://github.com/PowerBroker2/SerialTransfer/issues?q=is%3Aissue+spi)). Basically, the SPI API isn't consistent across all Arduino boards (which I find very, very frustrating), causing compile-time issues for several boards. I tried enabling SPITransfer.h for certain supported boards, but this didn't work out as well as I wanted it to. While I didn't want to remove SPI functionality since it does work for the supported boards, I decided to comment out the contents of that part of the lib. That way the small percentage of people who need SPITransfer.h can still use it with a little bit of elbow grease.

Assuming the SAMD21 cortex M0+ board is supported by SPITransfer.h, the compile-time error you're getting should go away if you comment the offending line out. That particular line is part of the documentation comments for that function.

K-Space909, 2021-03-16:
> That particular line is part of the documentation comments for that function.

I see it, and now I feel dumb. I think I've got it fixed up now. Thanks for the great support.

I'm not sure if this will work on the M0+, but I suspect it might. I think I just need to find equivalent statements for these lines, which I have confirmed will compile for the Uno but not for my Xiao boards:

    	SPCR |= bit(SPE);

and

    ISR(SPI_STC_vect){...}

Specifically, it looks like I don't have compiler definitions for SPCR or ISR

PowerBroker2, 2021-03-16:
Yeah, I had trouble researching the different APIs for SPI for the different boards. I'm going to close this comment for now, but if you do learn how to use SPI for your board, I'd be interested to know and you can comment that here.

closed 2021-03-16 by PowerBroker2

K-Space909, 2021-03-16:
Thanks man, I appreciate your help.