PowerBroker2/SerialTransfer #62: Does this work from an ATMEGA328 to a SAMD21?

mjbmikeb2, 2021-04-04:
A struct containing individual bytes transfers OK but a struct containing ints and floats get messed up.
The documentation doesn't mention inter-platform compatibility.

PowerBroker2, 2021-04-04:
The library works across platforms as long as the endianness and type sizes match. The MCUs you mentioned are both little endian, so that isn't a problem. However, the ATMEGA is an 8-bit MCU and the SAMD is 32-bit, meaning, if you say `int` without specifying a byte-size (i.e. `int16_t`), you will have parsing issues. My guess is you are using `int`, but using `int16_t` will solve your issue.

mjbmikeb2, 2021-04-05:
Turned out to be a padding issue. Adding
__attribute__((packed))
to the struct definition on the 32 bit end made it compatible with the serialised format of the 8 bit struct which doesn't do padding.
Depending on the layout of the struct, the padding on 32 bit platform can significantly increases the serialised size.
You might want to mention this in the documentation or examples.

closed 2021-04-05 by PowerBroker2