PowerBroker2/SerialTransfer #63: library makes i2c lcd flash

emmanuelsheshi, 2021-04-19:
good day sir, thanks so much for the library. but i noticed that it makes my i2c lcd flash. what can i do to resolve this? thanks in advance...

PowerBroker2, 2021-04-20:
I've used this library with one of my own I2C LCD displays and haven't had this problem. The issue is almost certainly with another part of your sketch.

closed 2021-04-20 by PowerBroker2

p1ngb4ck, 2022-12-23:
Sry to bother, but I think you forgot about the different types of displays here.
I have a 1602LCD myself used with SPI header board via SPI connection.
As soon as I use the line
"if(serialLine.available())" (where serialLine is myTransfer), my LCD goes completely blank. No idea why. There must be some side-effects on SPI, even though I didnt uncomment any SPI-header/functions of your lib