PowerBroker2/SerialTransfer #66: How can I pause serial transfer and then un-pause it ?

telemarketer, 2021-06-09:
So I am running a usb (UART) connection from my arduino(using the serialtransfer.h  library) to my pc (using the pyserial transfer library). 
This how each cycle of code goes:
I receive the list_ packet and save it to a list on the Arduino . Then the elements in the list are checked and used to control parameters of motor a using GPIO pins. Finally a return message is sent to acknowledge each transfer.

I was hoping to make the return message from Arduino as sort of a flag ; whether to wait or move onto the next transfer, this is on the python side of things. This is not the case as the sending of packets from Arduino is almost instantaneous even with delay functions

The python code is the standard code on your github with few changes to the parameters of  rx.obj function
here is the Arduino side of things
```
#include "SerialTransfer.h"

SerialTransfer myTransfer;
int motor1 = 13;
int motor2 = 10;

int obj_cordinates[9];
char reply [16] = {"Packet Recieved"};
void setup()
{
  Serial.begin(115200);
  pinMode(motor1,OUTPUT);
  pinMode(motor2,OUTPUT);
  
  myTransfer.begin(Serial);
}

void loop()
{
  if(myTransfer.available())
  {
    // use this variable to keep track of how many
    // bytes we've processed from the receive buffer
    uint16_t recSize = 0;
    uint16_t sendSize = 0;

    recSize = myTransfer.rxObj(obj_cordinates, recSize);
    sendSize = myTransfer.txObj(reply, sendSize);
    
     if( obj_cordinates[4] == 15) 
    {
      digitalWrite(motor1, HIGH);
      digitalWrite(motor2, LOW);
      
      delay(5000);
      
      digitalWrite(motor1, LOW);
      digitalWrite(motor2, HIGH);
      delay(5000);
    }

    myTransfer.sendData(sendSize);
  }
  
}
```

PowerBroker2, 2021-06-09:
> This is not the case as the sending of packets from Arduino is almost instantaneous even with delay functions

That sounds like the delays aren't being called, which seems more like a sketch issue. Try using debug prints via a second serial port (plus UART to USB converter) and see what exactly is going on.

Also note that you can use `sendDatum()` to replace `txObj()` and `sendData()` if and only if you're transferring a single object in a single packet.

telemarketer, 2021-06-10:
I am using an Arduino Uno and it doesn't have 2 serial ports. So while transferring via usb cable I can't use serial to print. 
For delays not working 

Its very wired that delays doesn't work as it baked into Arduino IC. To recap ***My only problem is this : delaying the*** 

```     }

    myTransfer.sendData(sendSize);
  } ```

***enough so that the program is run sequentially*** instead it runs as "if its on separate thread" and non sequentially .

PowerBroker2, 2021-06-10:
> So while transferring via usb cable I can't use serial to print.

That's not true, you can use softwareserial.h or another software uart library to communicate with a uart to usb converter.

> Its very wired that delays doesn't work as it baked into Arduino IC. To recap My only problem is this : delaying the `myTransfer.sendData(sendSize);` enough so that the program is run sequentially instead it runs as "if its on separate thread" and non sequentially .

If the `sendData()` calls are not being delayed, then it is most likely because your sketch is not calling `delay()` due to a bug somewhere in the sketch or Python.

telemarketer, 2021-06-10:
You were right the if statement wasn't being executed, there seems to an issue with Arduino processing the received list , I am sending a list a=[1,2,3,4,5,6,7,8,9] to the Arduino so then  a[5] > a[4] is always true for the list a , but this doesn't seem to be the case, also a[4] == 5 returns as false. 

Thank You for all the help so far :)

telemarketer, 2021-06-10:
> That's not true, you can use softwareserial.h or another software uart library to communicate with a uart to usb converter.
I think you need a usb to ttl converter for that, it would very helpful for debugging but the current corona situation in my country doesn't allow for non-necessity goods to restocked or delivered online.

telemarketer, 2021-06-10:
> That's not true, you can use softwareserial.h or another software uart library to communicate with a uart to usb converter.

 That's not true, you can use softwareserial.h or another software uart library to communicate with a uart to usb converter.
I think you need a usb to ttl converter for that, it would very helpful for debugging but the current corona situation in my country doesn't allow for non-necessity goods to restocked or delivered online.

PowerBroker2, 2021-06-10:
The reason you might be having issues is that Python normally defaults to ints being 32 bits wide while Arduino Uno ints are 16 bits wide.

telemarketer, 2021-06-11:
Changing the array to int16 using numpy breaks the listsize as it return None, because numpy.int16 arrays as classified as <class 'numpy.ndarray'> when using type() in python.

telemarketer, 2021-06-11:
Yes It is working Now!!!  Changing the ***int obj_cord[9]*** array to ***int32_t obj_cord[9]*** in the arduino code fixed it.

closed 2021-06-11 by telemarketer