PowerBroker2/SerialTransfer #68: Library Compiling Issue

AlexanderPeters, 2021-07-09:
I can't seem to get this library to compile. I've written up one of the simplest examples I could think of.

I keep getting this error message. 'class SerialTransfer' has no member named 'txBuff'

Are txbuff and rxBuff deprecated because it looks like they are defined on lines 53 and 54 of Packet.h so I don't understand what the issue is.

Please see the attached example and let me know if there's something I can change to get it to compile.

I am using 
Arduino IDE: 1.8.13
SerialTransfer Library:  3.1.1 (Installed through Arduino Library Manager)
Pop!_OS 20.10

[example.zip](https://github.com/PowerBroker2/SerialTransfer/files/6794305/example.zip)

PowerBroker2, 2021-07-10:
The problem is you're using an outdated example. Check out the updated examples [here](https://github.com/PowerBroker2/SerialTransfer/tree/master/examples). `rxBuff` and `txBuff` are now members of the `Packet` class, so the code should say `myTransfer.Packet.rxBuff[i]`.

AlexanderPeters, 2021-07-10:
Thanks, I should've picked up on that. I'll try it tomorrow to make sure that works. I desperately need to brush up on my C/C++ syntax.

AlexanderPeters, 2021-07-10:
Got it working. It should be noted that the fix required "packet" to start with a lower case. That way the Packet class object "packet" is referenced and not the Packet class itself.

The corrected line which fixed the compiling issue:
 myTransfer.packet.txBuff[i] = myTransfer.packet.rxBuff[i];

Attached is the corrected project file. 

Thanks for the help.
[example_fixed.zip](https://github.com/PowerBroker2/SerialTransfer/files/6796065/example_fixed.zip)

closed 2021-07-10 by AlexanderPeters

PowerBroker2, 2021-07-10:
> It should be noted that the fix required "packet" to start with a lower case

Oops, sorry about that! Glad you got it working