PowerBroker2/SerialTransfer #69: How to determine if a file is completely send?

DustinWatts, 2021-07-10:
Hi! 

Thanks for this great library! I have a question regarding UART though. On the TX side I am opening a file and sending it using this crude bit of code (ESP32) based on the uart_tx_file example:

```
#include "SerialTransfer.h"
#include "SPIFFS.h"

SerialTransfer myTransfer;

#define RXD2 16
#define TXD2 17

void setup()
{
  Serial.begin(115200);
  Serial1.begin(115200, SERIAL_8N1, RXD2, TXD2);

  // Launch SPIFFS file system  
  if(!SPIFFS.begin()){ 
    Serial.println("An Error has occurred while mounting SPIFFS");  
  }

  
  myTransfer.begin(Serial1);
}

void loop()
{
  char fileName[] = "dslr.jpg";
  File file2 = SPIFFS.open("/dslr.jpg");
  if(!file2){
      Serial.println("Failed to open file for reading");
      return;
  }
  uint16_t i = 0;
  const int fileSize = file2.size();
  char file[fileSize] = "";
  while(file2.available()){
     file[i] = file2.read();
     Serial.print (file[i]); //use for debug
     i++;
  }
  file[i] ='\0';
  // Serial.print (file); //use for debug
  file2.close();
    
    
  
  myTransfer.sendDatum(fileName); // Send filename
  
  uint16_t numPackets = fileSize / (MAX_PACKET_SIZE - 2); // Reserve one byte for current file index
  
  if (fileSize % MAX_PACKET_SIZE) // Add an extra transmission if needed
    numPackets++;
  
  for (uint16_t i=0; i<numPackets; i++) // Send all data within the file across multiple packets
  {
    uint16_t fileIndex = i * MAX_PACKET_SIZE; // Determine the current file index
    uint8_t dataLen = MAX_PACKET_SIZE - 2;

    if ((fileIndex + (MAX_PACKET_SIZE - 2)) > fileSize) // Determine data length for the last packet if file length is not an exact multiple of MAX_PACKET_SIZE-2
      dataLen = fileSize - fileIndex;
    
    uint8_t sendSize = myTransfer.txObj(fileIndex); // Stuff the current file index
    sendSize = myTransfer.txObj(file[fileIndex], sendSize, dataLen); // Stuff the current file data
    
    myTransfer.sendData(sendSize, 1); // Send the current file index and data
    delay(100);
  }
  delay(10000);
  Serial.print("File sent!");
}
```

Now on the RX side I am receiving all the data. But I want to also put that back in a file. How would I do that? How would I know if the end of the file is reached?

This is my code atm on the RX side (RP2040) based on the uart_rx_file example:

```
#include "SerialTransfer.h"

SerialTransfer myTransfer;

const int fileSize = 2000;
char file[fileSize];
char fileName[10];

#define RXD2 5
#define TXD2 4

int startTime;
int transferTime;

void setup()
{
  Serial.begin(115200);
  //Serial1.begin(9600, SERIAL_8N1, RXD2, TXD2);
  Serial2.setRX(RXD2);
  Serial2.setTX(TXD2);
  Serial2.begin(115200);
  
  myTransfer.begin(Serial2);
}

void loop()
{
  if (myTransfer.available())
  {
    if (!myTransfer.currentPacketID())
    {
      myTransfer.rxObj(fileName);
      Serial.println();
      Serial.println(fileName);
    }
    else if (myTransfer.currentPacketID() == 1)
      for(uint8_t i=2; i<myTransfer.bytesRead; i++)
        Serial.print((char)myTransfer.packet.rxBuff[i]);
    Serial.println();
  }
  
}
```

PowerBroker2, 2021-07-10:
After sending `fileName`, you could send a packet with an ID of 2 (or any other unique ID) and send the `numPackets` value. This means that whenever the `fileIndex` (first two payload bytes) is equal to the originally sent `numPackets` value, you know you're processing the last packet of the file

closed 2021-07-17 by PowerBroker2

YBeyin, 2021-09-24:
> After sending `fileName`, you could send a packet with an ID of 2 (or any other unique ID) and send the `numPackets` value. This means that whenever the `fileIndex` (first two payload bytes) is equal to the originally sent `numPackets` value, you know you're processing the last packet of the file

is it possible to determine from tx side? i want to check if struct received by rx.

PowerBroker2, 2021-09-24:
I'm not sure I understand. Determine what from the tx side? You could probably write the code so that after every packet sent by the TX side, it would wait some amount of time for an "ack" packet from the RX side and resend if the "ack" packet wasn't received by the TX side. Is that what you were asking?

YBeyin, 2021-09-24:
> I'm not sure I understand. Determine what from the tx side? You could probably write the code so that after every packet sent by the TX side, it would wait some amount of time for an "ack" packet [...]

yes this is what i am asking.

PowerBroker2, 2021-09-24:
Sorry, I don't

DustinWatts, 2021-09-24:
@PowerBroker2 I realized I never responded to this after your comment. I did something similar you suggested, and got something that works for me :)

Btw, @YBeyin You could, after sending the last packet, send a hash of the data you were sending and then on the RX side hash it, send it back an on the TX side compare those hashes.

YBeyin, 2021-09-24:
> @PowerBroker2 I realized I never responded to this after your comment. I did something similar you suggested, and got something that works for me :)
>
> Btw, @YBeyin You could, after sending the last [...]

some little example would be so great to understand how it does work. im pretty new at arduino :S thanks for advice btw.