PowerBroker2/SerialTransfer #70: Send picture from one ESP32CAM to another using SPI

Laggger164, 2021-07-14:
**TL;DR: All I want is to send the camera's frame buffer to another ESP32CAM. That's it. No streaming, just one photo every few minutes or so.**

If you think my approach could be different please tell me, I will gladly consider other options!

I am having a hard time figuring out how to do this, documentation is scarce presumably because most people don't use SPI. (I2C is way too convenient)
However, I found a forum thread which described a similar issue: [link](https://forum.arduino.cc/t/transfer-pictures-from-esp32-cam-to-esp32-via-serial/647488/29)

I wish to create a script that would take a picture on both ESP32CAM modules, send the picture from the first one to the second one and then have that one send both pictures to an FTP server. However, I want to use ethernet, since Wi-Fi seems to be really unreliable for some reason (and would like to use PoE) and the W5500 Ethernet modules communicate through SPI. 
So unless I can switch from I2C for the ESP to ESP communication to SPI for the Ethernet module... I must use SPI.

I am not very knowledgeable in serial communication so I was hoping someone could help me by giving me some guidance on how to do this. The thread I linked does help, but was written for I2C (correct me if I am wrong) and the code gets a little complicated and combined with the fact I can't find much documentation on this makes me a little uneasy when coding.

Thank you for any and all help!

Receiver code (very unfinished and not working, mostly a cut down example from mentioned thread)
```
/*Receives picture via serial and saves to ftp*/

#define DEBUG_ESP              //comment out to deactivate debug console

#ifdef DEBUG_ESP
  #define pDBGln(x) Serial.println(x)
  #define pDBG(x)   Serial.print(x)
#else 
  #define pDBG(...)
  #define pDBGln(...)
#endif

#include "SPITransfer.h"
SPITransfer myTransfer;
struct img_meta_data{
  uint16_t counter;
  uint16_t imSize;
  uint16_t numLoops;
  uint16_t sizeLastLoop;
} ImgMetaData;
uint16_t packetCounter=1;
uint16_t bufferPointer=0;
char tempImageBuffer[32000];

// FTP Client Lib
#include "ESP32_FTPClient.h"

// FTP Server credentials
char ftp_server[] = "";
char ftp_user[]   = ""; 
char ftp_pass[]   = "";

// picture name
String picPrefix ="";
String pic_name;  

// Connection timeout;
#define CON_TIMEOUT   10*1000                     // milliseconds

ESP32_FTPClient ftp(ftp_server, ftp_user, ftp_pass);

void setup(){
  //Serial.begin(Baud Rate, Data Protocol, Txd pin, Rxd pin);
  Serial.begin(115200);                                                     // Define and start serial monitor
  // 115200,256000,512000,962100

  SPCR |= bit (SPE);
  pinMode(MISO, OUTPUT);
  SPI.attachInterrupt();
  
  myTransfer.begin(SPI);
}

void loop(){
[... 243 lines cut ...]
    else
      pDBGln(F(" }"));
  }
}

void printBuf(){
  pDBG(F("Pixel Values: { "));
  for (uint16_t k=8; k<MAX_PACKET_SIZE; k++){
    pDBG(myTransfer.packet.txBuff[k]);
    if (k < (MAX_PACKET_SIZE - 1))
      pDBG(F(", "));
    else
      pDBGln(F(" }"));
  }
  pDBGln();
}

void stuffPixels(const uint8_t * pixelBuff, const uint16_t &bufStartIndex, const uint16_t &txStartIndex, const uint16_t &len){
  uint16_t txi = txStartIndex;
  for (uint16_t i=bufStartIndex; i<(bufStartIndex + len); i++)  {
    myTransfer.packet.txBuff[txi] = pixelBuff[i];
    txi++;
  }
}
```

PowerBroker2, 2021-07-15:
The original forum thread linked was about transmitting pictures via UART, not I2C. Also note the code in that post uses a slightly outdated API compared to the latest version.

The easiest thing to do would probably start with the [spi_tx_file.ino](https://github.com/PowerBroker2/SerialTransfer/blob/master/examples/spi_tx_file/spi_tx_file.ino) and [spi_rx_file.ino](https://github.com/PowerBroker2/SerialTransfer/blob/master/examples/spi_rx_file/spi_rx_file.ino) examples. It shouldn't be hard to replace the transmitted text file in the example with a picture file.

Sorry about the lack of formal documentation - this is something I struggle with for all my libraries. **However**, I make sure the source code is well organized, uses appropriate function/variable/constant names, and most importantly includes detailed doc string comments for each function. Basically, the documentation for the library is inside the library code itself. For API info, check the headers and for detailed "what does this function do" questions, check the cpp file comments.

Laggger164, 2021-07-15:
> The original forum thread linked was about transmitting pictures via UART, not I2C. Also note the code in that post uses a slightly outdated API compared to the latest version.
>
> The easiest thing to [...]

Thank you, I will try just the examples. I was taken aback by the complexity of the code in the thread I linked so I thought the examples were missing a lot.
I believe the user needed to make and send the packets individually. We are talking about kB of data after all.

Laggger164, 2021-07-15:
OK, looking at the example I devised something like this.
Right now there is a button at pin 4, which will pull the analog reading lower when pressed. A little problematic because there is the flash circuit on there, might figure out a better solution later.
Pushing the button executes the script that will take a picture, store it in the frame buffer with the function esp_camera_fb_get(), get its size using sizeof() and attempt to send it via SPIserial.
However I am not sure if sizeof() will give me what I want since there is some trickery involving pointers around it:
```
/**
 * @brief Obtain pointer to a frame buffer.
 *
 * @return pointer to the frame buffer
 */
camera_fb_t* esp_camera_fb_get();
```
Also, the receiver example script has a set constant for file size. I don't think the images will ever have a known constant size since there is compression and stuff going on. But I should be able to send the value over and get an acknowledgement from the other module right?

The sender code so far:
```
#include "esp_camera.h"
#include "Arduino.h"
#include "soc/soc.h"           // Disable brownout problems
#include "soc/rtc_cntl_reg.h"  // Disable brownout problems
#include "driver/rtc_io.h"
#include "SPITransfer.h"
//#include <EEPROM.h>            // read and write from flash memory
// define the number of bytes you want to access
//#define EEPROM_SIZE 1

#include <SPI.h>

const int buttonPin = 4;

bool buttonState = false;             // the current reading from the input pin

//unsigned long lastDebounceTime = 0;  // the last time the output pin was toggled
//unsigned long debounceDelay = 50;    // the debounce time; increase if the output flickers
 
RTC_DATA_ATTR int bootCount = 0;

SPITransfer myTransfer;

// Pin definition for CAMERA_MODEL_AI_THINKER
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27
#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22
 
int pictureNumber = 0;
long fileSize;
char fileName[] = "picture2.jpg";
  
void setup() {
  pinMode(buttonPin, INPUT);

  Serial.begin(115200);
  Serial.setDebugOutput(true);
  
  WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector
  digitalWrite(15, HIGH);
  
[... 95 lines cut ...]
              delay(100);
            }

            esp_camera_fb_return(fb);
            
            //delay(1000);
            
            // Turns off the ESP32-CAM white on-board LED (flash) connected to GPIO 4
            //pinMode(4, OUTPUT);
            //digitalWrite(4, LOW);
            //rtc_gpio_hold_en(GPIO_NUM_4);
          
            //esp_sleep_enable_ext0_wakeup(GPIO_NUM_13, 0);
           
            //Serial.println("Going to sleep now");
            //delay(1000);
            
            }
if (reading > 500) {
   buttonState = false;
   //Serial.println("state changed");
 }
  //lastButtonState = reading;
}
```

PowerBroker2, 2021-07-17:
“fb” is a pointer to a struct of type camera_fb_t. The source code for this struct can be found [in this header](https://github.com/espressif/esp32-camera/blob/4af111a5e274a2973fa10cd3c1ec69174a1bdb93/driver/include/esp_camera.h):

```C++
/**
 * @brief Data structure of camera frame buffer
 */
typedef struct {
    uint8_t * buf;              /*!< Pointer to the pixel data */
    size_t len;                 /*!< Length of the buffer in bytes */
    size_t width;               /*!< Width of the buffer in pixels */
    size_t height;              /*!< Height of the buffer in pixels */
    pixformat_t format;         /*!< Format of the pixel data */
    struct timeval timestamp;   /*!< Timestamp since boot of the first DMA buffer of the frame */
} camera_fb_t;
```

As you can see, you can call `fb->len` to determine the length of the total number of pixels in `fb->buf` instead of `sizeof()`. `sizeof()` will give you an incorrect answer anyway (it will give the length of the pointer type). Also, when sending pixel data, you will have to send data using `fb->buf` instead of `fb[<some_number>]`

Laggger164, 2021-07-20:
> “fb” is a pointer to a struct of type camera_fb_t. The source code for this struct can be found [in this [...]

Thank you for the clarification, I had the feeling something like this was in it.

I replaced sizeof() with fb->len as " fileSize = fb->len; "

There is a slight possibility that the code will work now.
Next however I need help with the sender code as I am not sure if there is already a function for it.

In the sender example [spi_tx_file.ino](https://github.com/PowerBroker2/SerialTransfer/blob/master/examples/spi_tx_file/spi_tx_file.ino) the fileSize is a set constant. However, picture sizes and their required memory will be variable. Is there a function that handles sending data about the size of the incoming file?

`
uint8_t sendSize = myTransfer.txObj(fileIndex); // Stuff the current file index
sendSize = myTransfer.txObj(fb[fileIndex], sendSize, dataLen); // Stuff the current file data

myTransfer.sendData(sendSize, 1); // Send the current file index and data`
I believe this sends data about an incoming packet yes?

In that case I should be able to just send fileSize too in a similar way. Just before the "for" loop begins.

Or is that even necessary? It could be possible to just send the packets, however many there are. But is there a system that would check for file integrity? Is is this outside the scope of this library?

Thank you for your help by the way!

PowerBroker2, 2021-07-20:
> In the sender example spi_tx_file.ino the fileSize is a set constant, same for the receiver example. However, picture sizes and their required memory will be variable. Is there a function that handles sending data about the size of the incoming file?

In this case, your filesize will be `fb->len` number of bytes.

> I believe this sends data about an incoming packet yes?

I'm not sure what you mean by that, but you do need to change 

```C++
sendSize = myTransfer.txObj(fb[fileIndex], sendSize, dataLen); // Stuff the current file data
```

to

```C++
sendSize = myTransfer.txObj(fb->buff[fileIndex], sendSize, dataLen); // Stuff the current file data
```

> In that case I should be able to just send fileSize too in a similar way. Just before the "for" loop begins.

Yes, you could do something similar much like how the example sends the filename before the for loop. Be sure to use a unique packet ID so that the receiver knows that packet carries file size information and not filename or file data.

Laggger164, 2021-07-20:
Right, thank you for the heads up, I definitely would have missed it.
 
> Yes, you could do something similar much like how the example sends the filename before the for loop. Be sure to use a unique packet ID so that the receiver knows that packet carries file size [...]

Awesome! I am going to send it as the first packet (number 0), this will also serve as a sort of confirmation that there is indeed a connection.

What about the file integrity? Its a lot of data, it might get corrupted at some point even though the connection is only around 15cm long. How could I deal with that?

PowerBroker2, 2021-07-20:
> What about the file integrity? Its a lot of data, it might get corrupted at some point even though the connection is only around 15cm long. How could I deal with that?

That's a really good question. The reason two bytes (a `uint16_t`) integer is reserved in each file data packet is to denote the `fileindex`. The receiver can use this `fileindex` value to see if any index is skipped (and hence corrupted). You could either pad the missing data with some value or devise a scheme that requests the missing data from the transmitter after the original file transmission is complete. This could get complicated fast, so doing lots of planning ahead of time and testing small will help a lot.

Laggger164, 2021-07-21:
> That's a really good question. The reason two bytes (a `uint16_t`) integer is reserved in each file data packet is to denote the `fileindex`. The receiver can use this `fileindex` value to see if [...]

Good to know! I will consider trying it once I get this up and running at all.

Modified to receive the fileSize instead of fileName, right now the fileName is set as a constant, later I will make a naming system.

Not sure if I have the right idea about storing the file data (array indexing in particular), how could I store the data in an array?

```
for(uint8_t i=2; i<myTransfer.bytesRead; i++) 
  { 
  Serial.print((char)myTransfer.packet.rxBuff[i]);
  file[i] = (char)myTransfer.packet.rxBuff[i];
  }
```
Current receiver code:

```
#include "SPITransfer.h"

SPITransfer myTransfer;

long fileSize;
char fileName[10];

volatile bool procNewPacket = false;

void setup()
{
  Serial.begin(115200);
  
  SPCR |= bit (SPE);
  pinMode(MISO, OUTPUT);
  SPI.attachInterrupt();
  
  myTransfer.begin(SPI);
}

void loop()
{
  if(procNewPacket)
  {
    procNewPacket = false;
    
    if (!myTransfer.currentPacketID())
    {
      myTransfer.rxObj(fileSize);
      Serial.println();
      Serial.println(fileSize);
      char file[fileSize];
      fileName = "pictureLeft";
    }
    else if (myTransfer.currentPacketID() == 1)
      for(uint8_t i=2; i<myTransfer.bytesRead; i++) 
      {
        Serial.print((char)myTransfer.packet.rxBuff[i]);
	file[i] = (char)myTransfer.packet.rxBuff[i];
        Serial.println();
      }
  }
}

ISR (SPI_STC_vect)
{
  if(myTransfer.available())
    procNewPacket = true;
}
```

The example from [this thread](https://forum.arduino.cc/t/transfer-pictures-from-esp32-cam-to-esp32-via-serial/647488/60) has a much more sophisticated receiving function, although I am not sure what changed since that old library version:

```
if(ImgMetaData.counter==1){  
      copyToImageBuff(MAX_PACKET_SIZE-sizeof(ImgMetaData));
    }else{
      if(ImgMetaData.counter==packetCounter){  
        if(packetCounter<ImgMetaData.numLoops){        
          copyToImageBuff(MAX_PACKET_SIZE-sizeof(ImgMetaData));
        }else if(ImgMetaData.counter==packetCounter){
          copyToImageBuff(ImgMetaData.sizeLastLoop);     
        }
      }
    }

void copyToImageBuff(uint16_t dataLenght){
  for(int y=0;y<dataLenght;y++){
    tempImageBuffer[bufferPointer+y] = myTransfer.rxBuff[y+sizeof(ImgMetaData)];
  } 
  bufferPointer+=dataLenght;
  packetCounter++;  
}
```

Current sender code:
I replaced the variable in sendDatum function to send the fileSize instead of fileName since naming will be handled by the receiver anyway. 

```
void loop() {
 // Take Picture with Camera
 int reading = analogRead(buttonPin);
 Serial.printf("Reading: %d\n", reading);
 uint16_t startIndex = 0;
 camera_fb_t * fb = NULL;
 
      if ((reading < 500) && (buttonState == false)) {
            buttonState = true;
            fb = esp_camera_fb_get();  
            if(!fb) {
              Serial.println("Camera capture failed");
              return;
            }
            Serial.println("Camera capture succeeded");

            fileSize = fb->len; //set variable fileSize

            myTransfer.sendDatum(fileSize); // Send fileSize
  
            uint16_t numPackets = fileSize / (MAX_PACKET_SIZE - 2); // Reserve one byte for current file index
            
            if (fileSize % MAX_PACKET_SIZE) // Add an extra transmission if needed
              numPackets++;
            
            for (uint16_t i=0; i<numPackets; i++) // Send all data within the file across multiple packets
            {
              uint16_t fileIndex = i * MAX_PACKET_SIZE; // Determine the current file index
              uint8_t dataLen = MAX_PACKET_SIZE - 2;
          
              if ((fileIndex + (MAX_PACKET_SIZE - 2)) > fileSize) // Determine data length for the last packet if file length is not an exact multiple of MAX_PACKET_SIZE-2
                dataLen = fileSize - fileIndex;
              
              uint8_t sendSize = myTransfer.txObj(fileIndex); // Stuff the current file index
              sendSize = myTransfer.txObj(fb->buff[fileIndex], sendSize, dataLen); // Stuff the current file data
              
              myTransfer.sendData(sendSize, 1); // Send the current file index and data
              delay(100);
            }

            esp_camera_fb_return(fb);
}
```

PowerBroker2, 2021-07-22:
> Not sure if I have the right idea about storing the file data (array indexing in particular), how could I store the data in an array?

That would almost work. You need to keep track of the `fileIndex` to know where the data in the current packet actually corresponds to in the file (since you're sending the data piecemeal)

Here's a way to do it:

```C++
myTransfer.rxObj(fileIndex);
myTransfer.rxObj(file[fileIndex], 2, myTransfer.bytesRead - 2);
```

For more info on `rxObj()` see:

https://github.com/PowerBroker2/SerialTransfer/blob/3367a6f2056ef2496906ff4bcb2c93f43720fd49/src/Packet.h#L112-L151

I think it's looking really good so far

Laggger164, 2021-08-03:
I am currently adding the Ethernet and FTP functionality to the receiver code, however there are a few problems when compiling.

The "ESP32_FTPClient.h" has been modified to work with Ethernet.h and EthernetClient.h instead of WiFi.h and WiFiClient.h respectively. You can find it [here](https://pastebin.com/u/Laggger164/1/Wk6Gkp8m)

Something is up with the ISR, SPCR, SPI.AttatchInterrupt and others, probably version incompatibility sooo... lots of work ahead.

Also file storage seems to be problematic to me, the compiler says is hasn't been declared in the scope of that function but I am not sure how else to make it. Any ideas?

The full code:
```
#include "SPITransfer.h"
#include "Arduino.h"
#include "esp_camera.h"
#include <Ethernet.h>
#include <EthernetClient.h> 
#include "ESP32_FTPClient.h"
#include "octocat.h"
#include <SPI.h>

SPITransfer myTransfer;

long fileSize;
char fileName;
uint16_t fileIndex;
bool serialTransferComplete;

volatile bool procNewPacket = false;

// Enter a MAC address for your controller below.
// Newer Ethernet shields have a MAC address printed on a sticker on the shield
byte mac[] = {
  0x00, 0xAA, 0xBB, 0xCC, 0xDE, 0x02
};

char ftp_server[] = "192.168.55.200";
char ftp_user[]   = "josai";
char ftp_pass[]   = "manager1";

String pic_name = "";

int files = 0;

camera_fb_t * fb = NULL;

// you can pass a FTP timeout and debbug mode on the last 2 arguments
ESP32_FTPClient ftp (ftp_server,ftp_user,ftp_pass, 5000, 2);

void setup()
{
  Serial.begin(115200);
  
  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer   = LEDC_TIMER_0;
  config.pin_d0       = 5;
  config.pin_d1       = 18;
  config.pin_d2       = 19;
  config.pin_d3       = 21;
  config.pin_d4       = 36;
  config.pin_d5       = 39;
  config.pin_d6       = 34;
  config.pin_d7       = 35;
  config.pin_xclk     = 0;
  config.pin_pclk     = 22;
  config.pin_vsync    = 25;
  config.pin_href     = 23;
  config.pin_sscb_sda = 26;
  config.pin_sscb_scl = 27;
  config.pin_pwdn     = 32;
  config.pin_reset    = -1;
[... 195 lines cut ...]

Receive_code:163:7: error: 'pic2_name' was not declared in this scope

       pic2_name += String( files ) + "left.jpg";

       ^

Receive_code:190:21: error: 'file' was not declared in this scope

       ftp.WriteData(file, sizeof(file)); //write transferred file to FTP server

                     ^

F:\Users\Jarko\Documents\Arduino\Receive_code\Receive_code.ino: At global scope:

Receive_code:203:5: error: expected constructor, destructor, or type conversion before '(' token

 ISR (SPI_STC_vect)

     ^

exit status 1

expected constructor, destructor, or type conversion before '(' token
```

closed 2022-01-27 by PowerBroker2

beckmx, 2023-02-14:
if you don't mind guys I have a question for everyone of you:
@Laggger164 why you didnt use serial(UART)? for logging purposes and how speedy was the implementation you did with SPI?

@PowerBroker2 I also did a similar project to this one (trasmitting images via your library) and it worked very good indeed, is SPI faster than UART?

greetings

Laggger164, 2023-02-16:
> if you don't mind guys I have a question for everyone of you: @Laggger164 why you didnt use serial(UART)? for logging purposes and how speedy was the implementation you did with SPI?
>
> @PowerBroker2 [...]

I believe at that time I thought I could only achieve the speeds necessary to send the image via SPI. UART might have been possible, but I didn't really try it.
Although I never managed to do it, I had to juggle other GPIO pins because of additional functionality and that would require turning off SPI and back on and basically it was a mess and I ended up going with a completely different solution.