PowerBroker2/SerialTransfer #71: Keen to understand the reasoning around default 50mS timeout

pauljeffress, 2021-07-15:
Hi there,

In Packet.h there is the following line

`const uint8_t DEFAULT_TIMEOUT = 50;`

I went looking as I was seeing the STALE PACKET error message when running my project.
I am using serialTransfer between an Adafruit Feather m4 express and a Sparkfun Artemis board.  
My code is based on the examples that tax and tax a Datum.
It was working fine when I was transferring back and forth a small struct with three chars and one integer. In fact in the Artemis to Feather direction that is still working fine.
But when I expanded the struct I send from the Feather to the Artemis to about 12 int16_t’s (24 payload bytes) I started seeing the Stale Packet message on the Artemis.

reading the library code I arrived at the “timeout” variable and found it defaults to 50mS but is configurable in the begin methods.  I changed it to 200mS temporarily to see if my larger strict would arrive intact at the Artemis and it was.

my question is, do you have any guidance around setting the timeout, or see any issues with taking it up to maybe a few 100mS?

I am running my link at 57600 by the way.

Thanks,
Paul

PowerBroker2, 2021-07-18:
There shouldn't be an issue with cranking the timeout to as high as you want. Timeouts and other errors can occur from datastream corruption or dropped bytes, which in turn can be caused by EMI, total loss of connection, or from buffer input overflow on the receiver side (not reading data faster than it arrives)

pauljeffress, 2021-07-18:
Cool, thanks.

closed 2021-07-18 by PowerBroker2