PowerBroker2/SerialTransfer #72: Using the ESP32-cam and sending images taken on it to something else.

AlexanderPeters, 2021-07-21:
This seems to be a fairly common project. (Either sending the image to another Arduino, a Raspberry Pi, or a PC) via one of the communication methods this library allows. Since the ESP32-cam is just so useful for how cheap it is. It looks like everybody (including myself) has been referencing the same old forum post with some confusion. And thus it seems like a good idea to have some example on how to do this as it's not trivial.

Aforementioned outdated forum post: 
https://forum.arduino.cc/t/transfer-pictures-from-esp32-cam-to-esp32-via-serial/647488/59

I'm currently working on ESP32-cam -> Raspberry PI over UART; so as soon as I get done with that code I'll be happy to contribute it if you see fit.

PowerBroker2, 2021-07-22:
Yeah, the API changed a little and the final code could probably use some improvements, so I definitely will document working code for this type of application as soon as someone (perhaps yourself?) can provide a working example. I don't own an ESP32-CAM to test with.

PowerBroker2, 2021-09-11:
No response, I assume you got it working

closed 2021-09-11 by PowerBroker2