PowerBroker2/SerialTransfer #73: payIndex not reset back to zero by the FSM

umaplehurst, 2021-08-04:
Consider a situation where a timeout occurs while in the `find_payload` state. The timeout handler will reset the FSM back to the `find_start_byte` state, no problems. However, `payIndex` is not reset back to zero by anything in the code (with the exception of when `payIndex == bytesToRec`). This means that RX is broken until enough further data arrives to cause `payIndex` to spill over and things get reset by overflow. I think that just resetting `payIndex` back to zero in `find_payload_len` when doing the state transition into `find_start_byte` is probably the right fix here?

PowerBroker2, 2021-08-06:
Thank you, this is a really good catch. You're right, `payIndex` isn't being reset in all cases. If I understand you right, the proposed fix is to put `payIndex = 0` as the first line in `case find_payload_len:`? I think that will fix it.

umaplehurst, 2021-08-06:
That would be fine; on my side, I just put `payIndex = 0` immediately before `state = find_payload` but what you propose would work just as well (payIndex only matters in the `find_payload` state so we only need to reset it on entry into the state). And thank you for all of your efforts in maintaining this great library!

PowerBroker2, 2021-08-06:
Awesome, I'll fix it asap. I'm glad you find the library(s) useful and always appreciate maintenance help/bug reports from the open source community!

PowerBroker2, 2021-08-06:
Fixed in [3.1.2](https://github.com/PowerBroker2/SerialTransfer/releases/tag/3.1.2)

closed 2021-08-06 by PowerBroker2