PowerBroker2/SerialTransfer #74: How i can use SerialTransfer with rs485 ?

YBeyin, 2021-09-10:
Hello everyone. I am trying to communicate 2 arduinos with rs485 (because of some long distance) how i can use SerialTransfer with rs485? I have read all examples, but can not find the solution. in uart , having an error  as "'Serial1' was not declared in this scope" . So is it not possible use SerialTransfer with rs485?

my rs485 and arduino connection is like :

// Arduino TX1 -> Rs485 DI
// Ardiuno RX0 -> Rs485 RO
// Arduino GND -> Rs485 GND
// Arduino 5V -> Rs485 VCC
// Arduino D2 -> Rs485 DE + RE

PowerBroker2, 2021-09-10:
SerialTransfer is compatible with RS485. The error you describe is not due to the RS485, but due to your Arduino not having a supported UART port other than USB. I'm guessing you're compiling for an Uno? Either way, SerialTransfer is also compatible with software serial libraries such as SoftwareSerial.h. Use one of those libraries to setup a software serial port to connect to RS485 and then pass that software serial port to SerialTransfer instead of Serial1.

YBeyin, 2021-09-10:
> SerialTransfer is compatible with RS485. The error you describe is not due to the RS485, but due to your Arduino not having a supported UART port other than USB. I'm guessing you're compiling for an [...]

i am using two arduino nano.one is for slave, other one is for master. i just plugged my rs485 as i described at first message. i did figure out nano does not support serial1, but still can not figure out how to do it. I have searched for examples but had nothing. rx for receiver isn't it?

PowerBroker2, 2021-09-10:
Something like this should work:

```C++
#include "SerialTransfer.h"
#include <SoftwareSerial.h>

SoftwareSerial mySerial(2, 3); // RX, TX
SerialTransfer myTransfer;

struct STRUCT {
  char z;
  float y;
} testStruct;

char arr[6];

void setup()
{
  Serial.begin(115200);
  mySerial.begin(9600); // changed baud from 115200 to 9600
  myTransfer.begin(mySerial);
}

void loop()
{
  if(myTransfer.available())
  {
    // use this variable to keep track of how many
    // bytes we've processed from the receive buffer
    uint16_t recSize = 0;

    recSize = myTransfer.rxObj(testStruct, recSize);
    Serial.print(testStruct.z);
    Serial.print(testStruct.y);
    Serial.print(" | ");

    recSize = myTransfer.rxObj(arr, recSize);
    Serial.println(arr);
  }
}
```

YBeyin, 2021-09-10:
> Something like this should work:
> ```

This is for slave (struck sender) right?

PowerBroker2, 2021-09-10:
That's for receiving, but the initial setup is the same for the transmitting arduino

YBeyin, 2021-09-10:
Okay thanks alot @PowerBroker2 i will try it. I hope so i can handle this. Your library is great. And i am so new at arduino.

YBeyin, 2021-09-11:
@PowerBroker2 forgive me but that did not work. I have been updated the main question with all details and codes. Still need some help.

PowerBroker2, 2021-09-11:
I need more context - what exactly is going wrong? Did you test the physical signal with a multimeter? Do you get a compiler error? How much % memory does the sketches use up? etc.

Also, I notice your rx sketch doesn't drive the RS485 direction pin low to receive.

YBeyin, 2021-09-11:
> I need more context - what exactly is going wrong? Did you test the physical signal with a multimeter? Do you get a compiler error? How much % memory does the sketches use up? etc.
> 
> Also, I notice your rx sketch doesn't drive the RS485 direction pin low to receive.

i am not having any compile error. even when i code the write the myTransfer.status on the screen that shows the value of status is 1.  %60 arround memory does the sketches uses.
with these codes i can send data to master and that is works but i want to send data as struct. thats why i am asking your help. 

here the codes that works with this connection: 

Codes for slave : 
```c++
//DHT22 Slave Ardiuno Nano

//Connections :

// ****** RS485 ********
// Arduino TX1 -> Rs485 DI
// Ardiuno RX0 -> Rs485 RO
// Arduino GND -> Rs485 GND
// Arduino 5V -> Rs485 VCC
// Arduino D2 -> Rs485 DE + RE

// ****** DHT22 ********
// Arduino D3 -> DHT22 Data
// Arduino GND -> DHT22 GND
// Arduino VIN -> DHT22 VCC

#include <DHT.h>
#include <SimpleTimer.h>

#define DHTPIN 3
#define DHTTYPE DHT22

DHT dht(DHTPIN,DHTTYPE); 
SimpleTimer timer;

float temp;
float hum;

void setup() {
  Serial.begin(9600);  
  dht.begin();
  timer.setInterval(2000L,getDhtValues);     
}

void loop() {  
  timer.run(); 
}

void getDhtValues(){
 temp=dht.readTemperature(); //get dht tempeture value
 hum=dht.readHumidity();   //get dht humidity value
 digitalWrite(2,HIGH);
  Serial.println(String(temp)); 
  delay(50);
  digitalWrite(2,LOW);
}
```

And the codes for master :

```c++
//Dht22 Master Arduino Nano With 20x4 Graphic Lcd

// Connections : 

// ****** RS485 *****
// Arduino TX1 -> Rs485 DI
// Arduino RX0 -> Rs485 RO
// Arduino D8 -> Rs485 DE + RE
// Arduino GND -> Rs485 GND
// Arduino 5V -> Rs485 VCC

// ***** LCD DISPLAY ****
// Arduino A4 -> Display SDA
// Arduino A5 -> Display SCL
// Arduino GND -> Display GND
// Arduino 5V -> Display VCC

#include <Wire.h>
#include <LiquidCrystal_I2cTUR.h>

int displayColumns = 20;
int displayRows = 4;
int enablePin = 8; 
String msg;
float temp;
LiquidCrystal_I2cTUR display(0x27, displayColumns, displayRows);  

void setup() {  
  Wire.begin();
  Serial.begin(9600); 
  display.init();                     
  display.backlight();
  pinMode(enablePin, OUTPUT);
}

void loop() {
 if(Serial.available()>0){
    msg=Serial.readString();      
    Serial.println(msg);      
  }
  display.clear();  
  display.setCursor(0,0);
  display.print("System Ready"); 
  display.setCursor(0,1);
  display.print(msg);   
  delay(1000);  
}
```

PowerBroker2, 2021-09-11:
Here's some example code for RS485 that I've used with SerialTransfer:

https://github.com/PowerBroker2/RND_Flight_Code/blob/26ac84d0507d0c558e5866226c323ce6e4ebb9b5/src/operational/Arduino/Ground_Station/RS485.h#L6-L25

Note that I'm using one port for TX and one for RX, but in both cases I drive the enable pin either high or low.

As for your code, I think I see the problem: You wired up the RS485 transceiver to the one hardware UART port instead of the softwareserial port. You can do either, but if you want to keep your wiring the same, you can delete all the softwareserial stuff and simply do this:

```C++
myTransfer.begin(Serial);
```

YBeyin, 2021-09-11:
> simply do this:
> 
> ```c++
> myTransfer.begin(Serial);
> ```

Thank you so much. You saved my whole project and my mental healt 👯 Thats works now. I am so new at arduino world. But loving this oop coding. Great library with so much potential.

closed 2021-09-11 by YBeyin