PowerBroker2/SerialTransfer #76: Code works on Nano BLE 33 but not on Mega 2560r3.

dbids, 2021-11-05:
**Describe the bug**
I was testing sample code on an Arduino Nano BLE33, where I got it to work fine.  After switching to the board I intend to use for this project, an Arduino Mega 2560 R3, I began seeing start bit errors in my python script receiving a string response back from the Arduino.  During debugging I played around with the delay a bit on both ends, and also encountered stop bit errors.  Are there any differences between the two boards that I should be aware of that would be leading to these issues?

**To Reproduce**
Steps to reproduce the behavior:
1. Send a packet containing a couple of ints to the Arduino.
2. Parse ints by using rxObj on the Arduino and then send back a string echoing the chosen function and arguments using txObj
3. See no error on one board, but much error on another.

**Expected behavior**
This is best illustrated in the screenshots below which depict the expected response string, and the erroneous response that I am now receiving.

**Screenshots**
Working Code on Arduino Nano BLE 33:
[image]

Broken code on Arduino Mega:
[image]
If I play around with the delay a little I get a stop bit error instead:
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version 20.04.3 LTS

**Additional context**
I can show my sample code if needed, though it is part of a greater project, so I would need to trim it down.

PowerBroker2, 2021-11-05:
Decode errors are usually because Python is expecting a different number of bytes than the packet has. When this happens, it's because your `txObj()` calls are formulated incorrectly. Check the byte length of each packet field in your Arduino sketch and verify your python script is expecting that exact number of bytes per field. `txObj()` is a powerful function, but only if used properly.

If you could strip down your code to just the communication pieces - while still producing the error - and post both simplified (and "runnable") programs, that will help

PowerBroker2, 2021-11-05:
Actually, since your code errors after changing MCUs, I know for a fact your `rxObj()` calls in Python are incorrect. If you use variable types without specifying bit lengths in your Arduino code (i.e. `int` vs `int16_t`), your compiler will assign different lengths for the same variable on different target processors. Since the Nano BLE 33 has a 32 bit processor, `int` will default to `int32_t` while `int` on the Mega will default to `int16_t`.

This variable length difference needs to be reflected in your Python code or specify the bit length in your Arduino code.

closed 2021-11-05 by PowerBroker2