PowerBroker2/SerialTransfer #78: Question: How do I implement this library with your ELMDuino library?

CrypticP, 2021-11-30:
I am new to coding so I am having some trouble implementing this library with some code using your ELMDuino library. I just need the output of RPM to transfer to my Arduino Uno to be used further.

Here is my code now :

`#include "BluetoothSerial.h"
#include "ELMduino.h"

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

//String MACadd = "DC:0D:30:48:8D:60";                         //enter the ELM327 MAC address
uint8_t address[6]  = {0x00, 0x1D, 0xA5, 0x22, 0xEC, 0x75};  //enter the ELM327 MAC address after the 0x

ELM327 myELM327;

uint32_t rpm = 0;

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  DEBUG_PORT.begin(115200);
  //SerialBT.setPin("1234");
  ELM_PORT.begin("ArduHUD", true);

  if (!ELM_PORT.connect(address))            //"OBDII" replaced by "address"
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    while (1);
  }

  if (!myELM327.begin(ELM_PORT))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }

  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.status == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else
  {
    printError();
  }
}

void printError()
{
  Serial.print("Received: ");
  for (byte i = 0; i < myELM327.recBytes; i++)
    Serial.write(myELM327.payload[i]);
  Serial.println();

  if (myELM327.status == ELM_SUCCESS)
    Serial.println(F("\tELM_SUCCESS"));
  else if (myELM327.status == ELM_NO_RESPONSE)
    Serial.println(F("\tERROR: ELM_NO_RESPONSE"));
  else if (myELM327.status == ELM_BUFFER_OVERFLOW)
    Serial.println(F("\tERROR: ELM_BUFFER_OVERFLOW"));
  else if (myELM327.status == ELM_UNABLE_TO_CONNECT)
    Serial.println(F("\tERROR: ELM_UNABLE_TO_CONNECT"));
  else if (myELM327.status == ELM_NO_DATA)
    Serial.println(F("\tERROR: ELM_NO_DATA"));
  else if (myELM327.status == ELM_STOPPED)
    Serial.println(F("\tERROR: ELM_STOPPED"));
  else if (myELM327.status == ELM_TIMEOUT)
    Serial.println(F("\tERROR: ELM_TIMEOUT"));
  else if (myELM327.status == ELM_TIMEOUT)
    Serial.println(F("\tERROR: ELM_GENERAL_ERROR"));

  delay(100);
}`

Thankyou so much for your help! I need this for my scholarship program and with limited online resources I am hoping you could give me a helping hand.

PowerBroker2, 2021-11-30:
I have an ESP32/Teensy 3.5 based car head's up display project with the source code [here](https://github.com/PowerBroker2/ArduHUD/tree/master/src). [The ESP32 sketch](https://github.com/PowerBroker2/ArduHUD/blob/master/src/ESP32_HUD/ESP32_HUD.ino) sends the RPM and MPH data to the Teensy 3.5 via SerialTransfer.h. Obviously, you'll need to change the [receiver sketch](https://github.com/PowerBroker2/ArduHUD/blob/master/src/Teensy_3x_HUD/Teensy_3x_HUD.ino) so that it's compatible with an Uno (the Uno does not have Serial1 - use Serial or SoftwareSerial.h).

CrypticP, 2021-12-01:
Awesome. Today I will give it all a try with that new code. For the ESP code I just need to change the connection to the ELM to be through a MAC address (simply because that's what has worked for me so far) and the teensy code from serial1 to serial.

CrypticP, 2021-12-01:
@PowerBroker2 I attempted to implement this into my ESP sketch. 

`#include "BluetoothSerial.h"
#include "SerialTransfer.h"
#include "ELMduino.h"

BluetoothSerial SerialBT;
#define DEBUG_PORT      Serial
#define LED_DRIVER_PORT Serial2
#define ELM_PORT        SerialBT

//String MACadd = "DC:0D:30:48:8D:60";                         //enter the ELM327 MAC address
uint8_t address[6]  = {0x00, 0x1D, 0xA5, 0x22, 0xEC, 0x75};  //enter the ELM327 MAC address after the 0x

const uint16_t MAX_RPM = 3500;
const uint16_t MIN_RPM = 700;
const uint8_t MSG_LEN = 16;

SerialTransfer myTransfer;
ELM327 myELM327;

enum fsm{get_speed, 
         get_rpm};
fsm state = get_rpm;

struct STRUCT {
  int8_t status;
  uint32_t rpm;
  float mph;
  char msg[MSG_LEN];
} carTelem;

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  DEBUG_PORT.begin(115200);
  LED_DRIVER_PORT.begin(115200);
  ELM_PORT.begin("ArduHUD", true);
  
  if (!ELM_PORT.connect(address))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner");
    while(1);
  }
  
  myTransfer.begin(LED_DRIVER_PORT);

  while(!myELM327.begin(ELM_PORT))
    delay(500);
}

void loop()
{
  switch(state)
  {
    case get_rpm:
    {
      float tempRPM = myELM327.rpm();

      carTelem.status = myELM327.status;
      memcpy(carTelem.msg, myELM327.payload, MSG_LEN);
      
      if(myELM327.status == ELM_SUCCESS)
        carTelem.rpm = tempRPM;
      else
        myELM327.printError();
      
      state = get_speed;
      break;
    }

    case get_speed:
    {
      float tempMPH = myELM327.mph();

      carTelem.status = myELM327.status;
      memcpy(carTelem.msg, myELM327.payload, MSG_LEN);
      
      if(myELM327.status == ELM_SUCCESS)
        carTelem.mph = tempMPH;
      else
        myELM327.printError();
      
      state = get_rpm;
      break;
    }
  }
  
  myTransfer.sendDatum(carTelem);
  
 
}`

This code worked great, fast connection to my ELM. 

The final problem I have ran into is understanding your receiver sketch. All I need is for a real-time rpm value to be constantly updated and transmitted to the Arduino Uno and printed to the Arduino Uno's serial monitor. I am so desperate to finish this final part. I am willing to send you an hour's pay of work for you to help me finish the code to upload to the Arduino Uno to achieve this.

CrypticP, 2021-12-03:
I will take help from anyone on this

CrypticP, 2021-12-05:
@PowerBroker2 I need your massive brain 😅😅😅

PowerBroker2, 2021-12-05:
What exactly do you need help understanding? Did you try and look at the source code, documentation, and examples?

CrypticP, 2021-12-05:
Genuinely, I have looked through examples, projects, and videos and cannot seem to understand why I can not connect the dots here. I have implemented your library into the code for the ESP32, but the code you sent from your HUD project was filled with so much I couldn't find/understand what code to use for the Arduino Uno to be able to receive values that the ESP32 is getting from my ELM327.

CrypticP, 2021-12-05:
I know the code is sitting right in front of my eyes which makes me look very ignorant and silly for commenting so much but it's for a passion project that I have been working on for a longtime; I just need the ESP32 to send real-time RPM values to the Arduino Uno through the serial monitor. Thankyou so much for commenting and helping me wrap my head around it.

PowerBroker2, 2021-12-06:
Here's what the Uno side should look like, but you will need to wire everything up properly and make sure the ESP32 is only sending one uint32_t value for RPM (my HUD project sends both RPM and speed):

```C++
#include "SoftwareSerial.h"
#include "SerialTransfer.h"

SoftwareSerial mySerial(2, 3); // RX, TX
SerialTransfer myTransfer;

uint32_t rpm;

void setup()
{
  Serial.begin(115200);
  mySerial.begin(9600);
  myTransfer.begin(mySerial);
}

void loop()
{
  if(myTransfer.available())
  {
    myTransfer.rxObj(rpm);
    Serial.print(rpm);
  }
}
```

closed 2021-12-06 by PowerBroker2

Doomcaliber12, 2021-12-07:
Hey @PowerBroker2 @CrypticP , i'm trying to replicate this project and also tested out your code but I ran into one issue. I tried to do the exact same thing but when I tried to adjust CripticP's code to make the esp send one uint32_t value for RPM, I ran into this error:

Arduino: 1.8.16 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None"

ESP_Code:87:3: error: 'myTransfer' does not name a type

   myTransfer.sendDatum(carTelem);

   ^~~~~~~~~~

exit status 1

'myTransfer' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Ive attached my code.
[ESP_Code.zip](https://github.com/PowerBroker2/SerialTransfer/files/7665230/ESP_Code.zip)

I'd really appreciate the help. Thank you
I just wanna mention, I was really impressed by your libraries. Good work

PowerBroker2, 2021-12-07:
That line of code is outside the scope of `loop()`. Perhaps you're missing a curly brace?

CrypticP, 2021-12-09:
yep @Doomcaliber12 that's definitely not in the loop. sadly it might not work either way. this is the error i just got using my code [image]

CrypticP, 2021-12-09:
my ELM was indicating a connection between my ESP through the LED's, but that error came out which is super weird because it's not in my code. @PowerBroker2 are those errors apart of the library ?