PowerBroker2/SerialTransfer #79: Variable

Neu59, 2021-12-15:
Hello friends thanks for this great tool.

I am using this code from the example.

"""#include "SerialTransfer.h"

SerialTransfer myTransfer;

struct STRUCT {
  char z;
  float y;
} testStruct;

void setup()
{
  Serial.begin(115200);
  Serial1.begin(115200);
  myTransfer.begin(Serial1);
}

void loop()
{
  if(myTransfer.available())
  {
    myTransfer.rxObj(testStruct);
    Serial.print(testStruct.z);
    Serial.println(testStruct.y);
  }
}"""
My level of programming is very basic. I am trying to extract the variable foat to use it in other functions of my code and I don't know how to do it.
Can you help me, thank you.

PowerBroker2, 2021-12-15:
For questions like this, please use the [Arduino Forum](https://forum.arduino.cc/)

closed 2021-12-15 by PowerBroker2

Neu59, 2021-12-15:
Thanks friend, sorry if it was an inappropriate question

PowerBroker2, 2021-12-15:
No worries, good luck!

Neu59, 2021-12-15:
I got it very simple

if(myTransfer.available())
  {
    myTransfer.rxObj(testStruct);
    Serial.print(testStruct.z);
    Serial.println(testStruct.y);
    variable=(testStruct.y);
  }
 float variable;    :+1: