PowerBroker2/SerialTransfer #81: Make requests to I2C slaves

TurboNerd95, 2022-01-21:
Hello, 

Very good job with the library. I had a lot of struggles interconnecting Arduino's when I started out and this eliminates the trouble.

I also think there should be a way for this library to incorporate the "Wire.requestFrom()" function from the Wire library, which allows a master Arduino to collect data from 1 or more I2C slave Arduino's, while also being able to transmit data to them. This would allow for 2 way communication, without the need to use multi-mastering (potentially unreliable).

Robin2 on the Arduino forum has a good example of this 2-way communication using the basic wire library; https://forum.arduino.cc/t/use-i2c-for-communication-between-arduinos/653958/4

Thanks

TurboNerd95, 2022-01-27:
Thank you for your interest in my feature request. I've been doing some tests with the "Wire.requestFrom()" function, where I try to send different sized packets of data back to the master. 

The weird drawback with using I2C requests is that the master will always fill it's receive buffer with the amount of bytes it requests, even if the slave ends up sending less bytes. I've tried to overcome this flaw by making a double request (1 request for the size of the data, and another request for the data itself), which you can read here;

https://forum.arduino.cc/t/i2c-optimization-of-the-wire-requestfrom-function-or-waste-of-time/951681  

No idea if this will actually increase the efficiency of the transmissions, but might be worth looking into.

fif911, 2023-01-22:
Hey @TurboNerd95. Were you able to make one master multiple slaves communication? That's what I need now, and I hope there is a way to use this lib for it.

TurboNerd95, 2023-01-25:
> Hey @TurboNerd95. Were you able to make one master multiple slaves communication? That's what I need now, and I hope there is a way to use this lib for it.

Yes, there is an additional argument in the sendData function (targetAddress), which allows the master to send to multiple slaves:

uint8_t I2CTransfer::sendData(const uint16_t& messageLen, const uint8_t& packetID, const uint8_t& targetAddress){

However, the "request" function I was asking about doesn't seem to be implemented yet.

fif911, 2023-01-28:
@TurboNerd95 Hey, thanks. I was able to discover that on the day I asked but now struggling to implement Slave to Master communication. (The same as `requestFrom` in the wire library). Any ideas if this is supported?

TurboNerd95, 2023-01-29:
> @TurboNerd95 Hey, thanks. I was able to discover that on the day I asked but now struggling to implement Slave to Master communication. Were you able to implement this request function somehow, or you used another solution? That's the same thing I also need now :)

I just ended up using the basic wire.h, in order to request data from the slave's. Here is a comprehensive set of examples: https://forum.arduino.cc/t/use-i2c-for-communication-between-arduinos/653958.

closed 2026-11-06 by PowerBroker2