PowerBroker2/SerialTransfer #86: Does this work with esp32?

camo200sx, 2022-02-27:
I get an error when using the tx/rx datum examples of

ERROR: STALE PACKET

regards

felipe-mello, 2022-02-27:
Hello! I've just used this library to communicate between an ESP 32 and a Teensy 4.0 and I had no problems whatsoever.

One thing I noticed, though, is that if you send a lot of that in a small amount of time, some is lost. For my application, I only need about 10 transactions per second and It's been working fine.

Unfortunately, I am still a beginner with the library, so I don't know exactly what is causing the error. If I must do an educated guess, I would say that there's a sync problem between your boards (must be that you're trying to send a lot of data at once, or the baud rates are not set correctly). Without the actual code, is kind of difficult to actually know what's going on. 

Best

felipe-mello, 2022-02-27:
Quick update: I also got the STALE PACKET error here. Turns out it was just a jumper bad connection. I moved the cables a little bit and everything started to work again... I guess checking your connections could also be a way to go.

PowerBroker2, 2022-02-28:
It does work with the ESP32, try using `struct __attribute__((packed, aligned(4)))` when creating structs to transfer

camo200sx, 2022-02-28:
Hmmm thanks for the help and suggestions. 

So I have further tested it. I was trying to get it going using HC-12 modules as they talk in serial but got the stale packet. Also tried with the esp's connected direct, same deal. I then changed the baud rate and it appears it is working @ 4800. I was originally using 1200 as this allows the HC12 to work at the longest possible range (which I need). I'm yet to try the HC-12 at above 1200 yet but direct serial connection is good at 4800

This then leads me to the question. Can it be done at 1200?

I have tried using `__attribute__((__packed__))` but no dice @ 1200. I'm quite the noob so unsure if using that correctly. Will post my code :)

**SEND**
```
#include <Arduino.h>
#include <Wire.h>
#include "SerialTransfer.h"

SerialTransfer myTransfer;

struct __attribute__((__packed__)) STRUCT
{
  char z;
  float y;
} testStruct;

void setup()
{
  Serial.begin(4800);
  Serial2.begin(4800);
  myTransfer.begin(Serial2);

  testStruct.z = '$';
  testStruct.y = 4.5;
}

void loop()
{
  myTransfer.sendDatum(testStruct);
  delay(500);
}
```

**RECEIVE**
```
#include <Arduino.h>
#include <Wire.h>
#include "SerialTransfer.h"

SerialTransfer myTransfer;

struct __attribute__((__packed__)) STRUCT {
  char z;
  float y;
} testStruct;

void setup()
{
  Serial.begin(4800);
  Serial2.begin(4800);
  myTransfer.begin(Serial2);
}

void loop()
{
  if(myTransfer.available())
  {
    myTransfer.rxObj(testStruct);
    Serial.print(testStruct.z);
    Serial.println(testStruct.y);
  }
}
```

nomakewan, 2025-04-25:
Hey there, I know this is an old issue but since I have a little experience with the ESP32 and SerialTransfer (and dealt with this issue this week) I wanted to make sure this issue could be closed.

The key here is that the "Stale Packet" error is only thrown if the end of a data packet is not read out of the buffer before the timeout, which in the default constructor for SerialTransfer is 50ms. At 1200 baud it takes about 8ms to transfer 1 byte of data. Your test struct is 5 bytes of data, and there are 6 bytes of overhead from the SerialTransfer encapsulation, for a total of 11 bytes. This means it would take 88ms to transfer your struct, which of course is larger than 50ms.

As you saw, increasing the baud rate helped; at 4800 it takes about 2ms to transfer 1 byte, so your test struct would take 22ms, which is less than half the timeout and plenty safe.

You need to be aware of how long it will take to transfer, be aware that there is overhead reading from the serial buffer, and then account for that in setting a new timeout value (or increasing your baud rate, whichever is appropriate for your setup).

To increase the timeout for SerialTransfer, declare your `begin` call as follows:

`myTransfer.begin(Serial2, false, Serial, 150);`

This sets the serial port SerialTransfer will use to `Serial2`, will set debug logging to `false`, will set the (now unused) debug console output to `Serial`, and finally sets the timeout for data packets to `150` milliseconds, plenty of time to receive 88ms worth of data transfer.

Hope that helps!

closed 2025-07-14 by PowerBroker2