PowerBroker2/SerialTransfer #87: More of a question, I think...

MacroPc, 2022-03-01:
New to Arduino but have years of VB6.0 experience (not that its all that similar...). Anyways, using I2C send/rec datum sketches I get:
ERROR: STOP_BYTE_ERROR

If I use the following. At first I thought I was exceeding the 254byte limit but then looking at the values and each variable in the struct I realize I'm not even close.

Rec code:

`#include "I2CTransfer.h"

I2CTransfer myTransfer;

struct STRUCT {
//  int16_t FillScreen;
  //char strText;
 // int8_t TextSize;
 // uint16_t TextColor;
 // uint16_t BColor;
//  int16_t TextX;
//  int16_t TextY;
  int16_t ColorToUse;
  bool DrawLine;
  bool FastLineH;
  bool FastLineV;
 // int16_t LineWidth;
 // int16_t LineHeight;
  bool DrawPixel;
  bool DrawCircle;
  bool DrawTriangle;
  int8_t CircleRadius;
 // int16_t CircleThickness;
  bool CircleSolid;
  bool DrawRect;
  bool SolidTriangle;
  //int16_t RectWidth;
 // int16_t RectHeight;
  int8_t RectRadius;
  bool RectSolid;
  bool RoundRect;
  int16_t X0;
  int16_t X1;
  int16_t Y0;
  int16_t Y1;
  int16_t X2;
  int16_t Y2;
} testStruct;

/////////////////////////////////////////////////////////////////// Callbacks
void hi()
{
  myTransfer.rxObj(testStruct);
  //Serial.println(testStruct.FillScreen);
  //Serial.println(testStruct.TextX);
 // Serial.println(testStruct.TextY);
  Serial.println(testStruct.ColorToUse);
  Serial.println(testStruct.DrawLine);
  Serial.println(testStruct.FastLineH);
  Serial.println(testStruct.FastLineV);
  Serial.println(testStruct.DrawPixel);
  Serial.println(testStruct.DrawCircle);
  Serial.println(testStruct.DrawTriangle);
  Serial.println(testStruct.CircleRadius);
  Serial.println(testStruct.CircleSolid);
  Serial.println(testStruct.DrawRect);
  Serial.println(testStruct.SolidTriangle);
  Serial.println(testStruct.RectRadius);
  Serial.println(testStruct.RectSolid);
  Serial.println(testStruct.RoundRect);
  Serial.println(testStruct.X0);
  Serial.println(testStruct.Y0);
  Serial.println(testStruct.X1);
  Serial.println(testStruct.Y1);
  Serial.println(testStruct.X2);
  Serial.println(testStruct.Y2);
  
}

// supplied as a reference - persistent allocation required
const functionPtr callbackArr[] = { hi };
///////////////////////////////////////////////////////////////////

void setup()
{
  Serial.begin(115200);
  Wire.begin(0);

  ///////////////////////////////////////////////////////////////// Config Parameters
  configST myConfig;
  myConfig.debug        = true;
  myConfig.callbacks    = callbackArr;
  myConfig.callbacksLen = sizeof(callbackArr) / sizeof(functionPtr);
  /////////////////////////////////////////////////////////////////
  
  myTransfer.begin(Wire, myConfig);
}

void loop()
{
  // Do nothing
}`

SEND CODE:

`#include "I2CTransfer.h"

I2CTransfer Transfer;

struct STRUCT {
 // int16_t FillScreen;
  //char strText;
 // int8_t TextSize;
 // uint16_t TextColor;
 // uint16_t BColor;
//  int16_t TextX;
 // int16_t TextY;
  int16_t ColorToUse;
  bool DrawLine;
  bool FastLineH;
  bool FastLineV;
 // int16_t LineWidth;
 // int16_t LineHeight;
  bool DrawPixel;
  bool DrawCircle;
  bool DrawTriangle;
  int8_t CircleRadius;
 // int16_t CircleThickness;
  bool CircleSolid;
  bool DrawRect;
  bool SolidTriangle;
  //int16_t RectWidth;
 // int16_t RectHeight;
  int8_t RectRadius;
  bool RectSolid;
  bool RoundRect;
  int16_t X0;
  int16_t X1;
  int16_t Y0;
  int16_t Y1;
  int16_t X2;
  int16_t Y2;
} testStruct;

void setup()
{
  Serial.begin(115200);
  Wire.begin();

  Transfer.begin(Wire);

 // testStruct.FillScreen = 10000;
 // testStruct.TextX = 66;
//  testStruct.TextY = 100;
  testStruct.ColorToUse = 11111;
  testStruct.DrawLine = true;
  testStruct.FastLineH = false;
  testStruct.FastLineV = true;
  testStruct.DrawPixel = true;
  testStruct.DrawCircle = false;
  testStruct.DrawTriangle = false;
  testStruct.CircleRadius = 5;
  testStruct.CircleSolid = true;
  testStruct.DrawRect = false;
  testStruct.SolidTriangle = false;
  testStruct.RectRadius = 5;
  testStruct.RectSolid = false;
  testStruct.RoundRect = true;
  testStruct.X0 = 222;
  testStruct.Y0 = 411;
  testStruct.X1 = 323;
  testStruct.Y1 = 256;
  testStruct.X2 = 255;
  testStruct.Y2 = 254;
 
}

void loop()
{
  Transfer.sendDatum(testStruct);
  delay(5000);
}`

Now If I comment out the top variable in the struct (ColorToUse), it works fine. If I leave it in I get the error. That's that issue. My question part of this (besides what I posted previously), is if it would be possible to set up a method of sending and receiving separate structs.

My project is a "shield" for a 3.0" screen I have. It uses 16 pins (i think) and thought it would be useful to have one arduino tell another one what to display on the screen. The previous method I tried using to communicate wasn't able to send the different types of variables accurately; yours does. It would be nice if I could create a struct for each type of graphic I want to display (lines, circles, text, etc....). Having this "shield" I could add to numerous future projects with just 2 wires would be a real asset in experiments and learning.

So if there a problem with my current code or would it be possible to accomplish the ladder?

PowerBroker2, 2022-03-03:
The callback functionality seems to be a little unstable for some reason, I'd suggest you process the data normally. That being said, note that you need to call ` myTransfer.tick();` in the `loop()` for any processing to happen on the RX side.

MacroPc, 2022-03-03:
I get "No member named 'tick' in 'I2CTransfer'clang(no_member)" in the new RC build 2.0.0 when adding myTransfer.tick().

I'm trying your data examples but I'm getting the same crc error even after removing half the data I'm trying to send. When using the exact structs posted above that gave me the BYTE STOP error I still get the CRC error in the data examples. Did I break it?

I will try the other protocols you have examples for and see if I can get those to work.

I don't care what protocol I use or what the setup is ( as long as its less wires used than whats required by the screen I have. In the future I'm going to stick with SPI displays), just as long as I can get this to work. I didn't think I would have this much trouble simply sending some variables!

PowerBroker2, 2022-03-03:
Oh, I thought you were using UART, oops. Are you using the latest version of the library?

PowerBroker2, 2022-03-03:
Note that there is a maximum buffer length for Arduino's implementation of I2C: [see here](https://arduino.stackexchange.com/a/14379/61572)

MacroPc, 2022-03-03:
Yes I just downloaded it 2 days ago. If I was maxing out the buffer size I would think that it would work when I removed some variables from the struct (including the rest of the code ofcourse). But I still get the errors. Ive tried to pinpoint which of the variables/values is causing the code not to work but I haven't pinpointed it out yet. I had it working sending all but (maybe 5?) of the variables. So removing those and more you would think it would work again. I kept thinking there was a typo somewhere where the structs didn't equal eachother but I've re-wrote them numerous times and copy/pasted them into the other window.

PowerBroker2, 2022-03-03:
Are you able to get the I2C examples working? Are you sure your wiring is good with connected grounds?

TurboNerd95, 2022-03-17:
The Arduino Wire library allows for each I2C transmission to be a max of 32 bytes.

This means that the 254 packet limit is only achievable with multiple I2C transmissions.  

by a quick count, the size of your struct (excluding the commented out variables) seems to be about 27 bytes. 

However, the SerialTransfer library also contains packet specific bytes, such as the start byte, PacketID, end byte ect, which is also sent with the transmission. So I'm assuming that this will bring the total bytes to over 32. 
 
@PowerBroker2 , what is the maximum amount of usable data we can include in a single I2C transmission, when taking into account these packet specific bytes?

closed 2022-05-14 by PowerBroker2