PowerBroker2/SerialTransfer #89: Is there an error in the file transfer example?

yursky91, 2022-03-27:
Hi!

I don't quite understand the values used in the [uart_tx_file.ino](https://github.com/PowerBroker2/SerialTransfer/blob/master/examples/uart_tx_file/uart_tx_file.ino) file.
Why we use `(MAX_PACKET_SIZE - 2)` in some places, and just `MAX_PACKET_SIZE` in others.
In particular, the calculation of `numPackets` seems to be going wrong:
```c
uint16_t numPackets = fileSize / (MAX_PACKET_SIZE - 2); // Reserve one byte for current file index
if (fileSize % MAX_PACKET_SIZE) // Add an extra transmission if needed
  numPackets++;
```
This comment is also not very clear: `// Reserve one byte for current file index` 

Same here:
```c
uint16_t fileIndex = i * MAX_PACKET_SIZE; // Determine the current file index
uint8_t dataLen = MAX_PACKET_SIZE - 2;

if ((fileIndex + (MAX_PACKET_SIZE - 2)) > fileSize) // Determine data length for the last packet if file length is not an exact multiple of MAX_PACKET_SIZE-2
  dataLen = fileSize - fileIndex;
```
What does `-2` mean?

mickeyvanolst, 2022-04-14:
@yursky91 did you end up figuring out how to get this to work?

PowerBroker2, 2022-04-15:
The code itself should be more or less ok as is, but there is a typo in the comment - I reserve "two" bytes for the current file index, not just one. That's where the `-2` comes from