PowerBroker2/SerialTransfer #90: When transmitting packets larger than 32 bytes got ERROR STALE PACKET

janczeresnia, 2022-04-27:
Hi
Is it possible to send more data than 32 bytes in one packet via hardware UART between two microcontrollers without error "ERROR STALE PACKET" ?
I use Rpi Pico and Seed Studio Xiao.

PowerBroker2, 2022-04-27:
You can send up to (and including) 254 bytes of data per packet

janczeresnia, 2022-04-27:
I know, I read the Packet.h header, but when I send a packet larger than 32 bytes, the STEEL PACKET error appears on the receiver side. It needs to send a 90-byte packet in one transmission. I have no idea how to fix this problem. I have tried different packet sizes below 32 bytes it works fine. Approximately every fifth packet passes between 33 and 48 bytes. Nothing goes up from 48 bytes. I tried different UART speeds, it doesn't change anything.

Thank you for the quick reply.

closed 2022-04-28 by janczeresnia