PowerBroker2/SerialTransfer #92: Does it work with ESP8266(NODEMCU V3)?

Senior97, 2022-05-12:
I tried example code named uart_rx_data and uart_tx_data with NodeMCU v3 but it didnt work. Does it work with NodeMCU ?

PowerBroker2, 2022-11-14:
If the MCU you're using doesn't have the wire library in it's core, you should probably poke around the forums (either Arduino or Espressif). If it continues to be an issue, you could probably comment out all of the files/includes that are associated with Wire.h

wassx, 2022-11-14:
Ok, I assume the answer is in here: https://www.arduino.cc/reference/en/language/functions/communication/wire/
NodeMCU is not supported with the `Wire.h` library

wassx, 2022-11-14:
@PowerBroker2 ok thanks for the quick reply, I will use UART for comms and comment the I2C out then.

PowerBroker2, 2022-11-14:
No prob. I like having I2C and SPI capabilities built into this library, but it's annoying when several board core libraries have inconsistent I2C/SPI APIs causing unnecessary problems with this lib. I probably should at some point put in preprocessor directives to automatically exclude the I2C portion of the lib if compiling for NodeMCU, but I'm busy atm

closed 2022-11-14 by PowerBroker2