PowerBroker2/SerialTransfer #93: Please provide examples of transferring more types of data.

ipsod, 2022-08-30:
Just like you provide examples for spi, i2c, etc., please provide examples for all possible data types.  Or, if too much is possible, then just some basics (int and uint 8, 16, 32 bit, float, double, byte array).

PowerBroker2, 2022-08-30:
I'm not doing all possible datatypes. As long as the datatypes match between sender and receiver, you should be good. Do you have any specific questions?

ipsod, 2022-08-30:
I think I learned what I need by looking at the code and comments for rxObj.  I was trying to lazily evaluate the library last night, to figure out if it would do what I wanted, and didn't understand how encoding/decoding was happening.

Thanks for the library.  Looks like it's just what I wanted.

closed 2022-08-30 by ipsod