PowerBroker2/SerialTransfer #94: Packet reset() overrides meaningful status in parse()

RayNieport, 2022-10-04:
**Describe the bug**
In Packet.cpp, the state machine in the `parse()` function calls `reset()` right before returning `bytesRead` in several locations.
Since `reset()` sets `status = CONTINUE`, the more meaningful status set by `parse()` is lost.

**Expected behavior**
For example - if the calculated CRC is different from the one received, the expected behavior is that `parse()` will return `bytesRead = 0`.
Then, we should be able to check and see that `SerialTransfer.packet.status == CRC_ERROR`. However, because `reset()` is called, we instead find that `SerialTransfer.packet.status == CONTINUE`.

**To Reproduce**
1. Provide an input/RX packet with incorrect CRC value
2. Call `packet.parse()`
3. Check `packet.status`

**Additional context**
After each new byte is received and parsed, I set `SerialTransfer.status = SerialTransfer.packet.status`. My intention is to then `sendDatum(SerialTransfer.status)` as a response to the original sender once the entire packet has been received. That way, if `SerialTransfer.status < 0`, the original sender will know to resend the packet since there was some error in transmission. It might also be advisable to make all "error" statuses in Packet.h negative numbers (just to stick with convention).

PowerBroker2, 2022-10-05:
I removed the `packet.status` update in the `reset()` function. It compiles fine for me and I don't expect it to break anything. Can you verify the master branch solves this issue? Thanks!

RayNieport, 2022-10-05:
Seems to have done the trick. I was a little worried it might break something but so far I haven't seen anywhere that it causes issues. Thank you!

Gotta love issues that have a nice easy fix  :)

Super useful lib by the way, I've been using it alongside pySerialTransfer to send files to/from a RPi Pico.

closed 2022-10-05 by RayNieport

PowerBroker2, 2022-10-05:
Glad you find it useful and thanks for the bug id!