PowerBroker2/SerialTransfer #95: Best method for checking that a packet is valid before reading the contents?

jessestevens5b, 2022-10-19:
I've been integrating this library into a remote control for a robot so I can have 2 way communication, but I'm find the documentation a little thin on the ground for a particular feature.

I'm not using the callback method, just the method shown in one of the examples (via UART) with the .available method, then using the .rxObj method to read the received data into a buffer array to decode.

It's not clear to me however whether this means it will only do this if a successful and CRC-checked packet in the correct order has been received, or I need to read a flag from somewhere before trusting that this data and decoding it?

I've had a quick read through the library source and couldn't get much clarification, so thought I'd quickly ask here before I delve any deeper. Perhaps if possible a bit more detail in the examples could be good? (I'm happy to contribute once I understand what is functionally happening)

PowerBroker2, 2022-10-19:
The `available()` member function only returns true if a packet has been successfully parsed (including vetting with CRC, etc). You can double check if you really want to by testing the `packet.status` flag. Here are all the possible status values:

https://github.com/PowerBroker2/SerialTransfer/blob/06804748ba26fb5adc8e3db436016caafa0681a6/src/Packet.h#L22-L28

jessestevens5b, 2022-10-19:
ah brilliant! Then it looks like what I've been doing is working.

My follow on question from this then, is that I'm running a super tight loop where timing is pretty critical, and I see that there is a timeout config option with a default of 50. Does this mean that every time a packet comes in it will wait for 50ms blocking until then? I'm just noticing some lag and trying to track down where it might be coming from and wondering where I can shave off any kind of blocking that has possibly been introduced in using this library.

PowerBroker2, 2022-10-20:
The `available()` function has a while loop that processes all bytes present in the serial input buffer, but that shouldn't block your code for very long whatsoever. The timeout parameter is to determine when a packet becomes "stale":

https://github.com/PowerBroker2/SerialTransfer/blob/06804748ba26fb5adc8e3db436016caafa0681a6/src/Packet.cpp#L121

Imagine you receive half of one packet, some time passes, and then a new packet starts. Instead of parsing data until you get a payload length, CRC, or stop byte error, the lib says "it's been way too long since the parsing of this packet started, therefore this data is probably from a new packet, let's just restart the parsing with this new data."

I have had issues with what I perceived to be latency issues like you mention, but was easily fixed by adding a ~10ms delay between transmits. After much thought, I came to the conclusion that the issue was not latency but was most likely serial input/output buffer overflows (too much data way too fast with small buffers). If your code is not the source of the latency and you think it's the lib - try increasing your serial input/output buffer sizes to >=255 bytes (or whatever works for you). There should be online tutorials on how to do this.

jessestevens5b, 2022-10-20:
This is great thanks for being so responsive. Yep definitely on top of the spacing of packets (I'm transmitting via XBee wireless UART links), as I know it's not great to overblow buffers like that. I'm thinking maybe it's just some lost bytes along the way with poor reception causing it to wait for timeouts, so I'll try out shrinking this time out so that each time it happens we don't wait for the default length of time.

Thanks!

closed 2022-10-20 by PowerBroker2